fmartinou/teleinfo2mqtt #9: Conteur en échec

ckd77, 2021-07-28:
Bonjour le toulousain ;)

Sur un pi 0, avec un modem GCE en USB sous /dev/ttyUSB0.
Je lance le conteneur, mais ce coupe. 

```
CONTAINER ID   IMAGE                     COMMAND                  CREATED          STATUS          PORTS     NAMES
b4ac8eeb53de   fmartinou/teleinfo-mqtt   "/usr/bin/entrypoint…"   38 seconds ago   Up 14 seconds             teleinfo-mqtt
```

```
root@raspberrypi:/home/teleinfo# docker ps -a
CONTAINER ID   IMAGE                     COMMAND                  CREATED              STATUS                        PORTS     NAMES
b4ac8eeb53de   fmartinou/teleinfo-mqtt   "/usr/bin/entrypoint…"   About a minute ago   Exited (132) 52 seconds ago             teleinfo-mqtt
```

```
# docker logs teleinfo-mqtt

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbefb65c4

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbeebb5b4
Illegal instruction (core dumped)
Illegal instruction (core dumped)

#
# Fatal error in , line 0
#

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbec705c4
unreachable code
#
#
#
#FailureMessage Object: 0xbe9275b4
Illegal instruction
Illegal instruction (core dumped)

#
# Fatal error in , line 0
#

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbefa95c4
unreachable code
#
#
#
#FailureMessage Object: 0xbed8c5b4
Illegal instruction (core dumped)
Illegal instruction (core dumped)
```

Via un script python, pas de soucis, je relève bien les infos du modem. Aurais tu une piste? Le but est d'utiliser ton conteneur et intégrer a mon broker sous HA comme tu fais tout simplement et proprement :) En te remerciant.

fmartinou, 2021-07-29:
Salut,

N'ayant moi-même pas de pi zéro sous la main, je n'ai pas pu testé cette Arch directement.
Je suis sûr que l'image a fonctionné sur Pi Zéro puisque certains utilisateurs le font tourner dessus.

Il se peut que ça soit du à une montée de version de node.js ou d'une dépendance :/
(je suis notamment passé de nodejs 14 à nodejs 16 récemment ; c'est la version taggée `4.0.0`)

Peux-tu essayer la version `fmartinou/teleinfo-mqtt:3.1.0` ou `fmartinou/teleinfo-mqtt:3.0.0` ?

ckd77, 2021-07-29:
Super merci impec avec la version 3.1.0 :)

fmartinou, 2021-07-29:
Super !
Maintenant, il va falloir que je trouve ce qui provoque la régression en 4.0.0 🤔

fmartinou, 2021-08-05:
J'ai publié une nouvelle version **`5.0.0`**.

J'ai rollbacké la version de node.js de 16 vers 14 car il se peut que ça soit ce changement-là qui fait planter l'appli sur les Pi Zéro.

Et quelques breaking changes sur cette version par contre ; tout est expliqué dans la doc mise à jour pour l'occasion :
https://fmartinou.github.io/teleinfo-mqtt

Si tu as l'occation de tester (je n'ai pas de Pi zéro à dispo ;)
Merci pour ton retour.

ckd77, 2021-08-05:
Coucou toi 

Je suis en congé à 1h de chez toi dans le 31 ^^

Sa attendra mon retour ;)

fmartinou, 2021-08-05:
Profite bien alors.
(et que la météo soit avec toi ;)

ckd77, 2021-08-12:
Coucou!

Dernier jour de vacs je rentre demain mais temps moche là, j’ai donc profité de tester ;) Et la version 5.0.0 fonctionne parfaitement sur pi 0 ;) 
Merci a toi.

fmartinou, 2021-08-12:
Parfait !
Ça veut donc dire que le pi zéro ne supporte pas Node.js 16.
Tant pis :(

closed 2021-08-12 by fmartinou

tms0, 2023-01-11:
Hello @fmartinou,

Je suis tombé sur un problème similaire, et j'ai trouvé l'explication et la solution ici : https://github.com/Koenkk/zigbee2mqtt/issues/7662

En montant de version l'image, tu es passé sur une version plus récente d'Alpine, qui nécessite d'installer la version backportée de libseccomp2 sur le host si ce dernier est une Debian Buster...!