fmartinou/teleinfo2mqtt #10: Hass identifier n'apparait pas dans le topic

WLCD, 2021-07-31:
Bonjour,

Je viens de déployer la version 3.1.0 sur un raspi zero (car la 4.0.0 ne veut pas s'exécuter) et j'ai noté que l'identifiant que j'ai défini dans le conteneur (linky) n'apparait pas dans le topic mqtt 

[image]

Et voici l'ouput coté mqtt

[image]

Le tableau JSON avec les valeurs relevées est bien transmis au broker 

[image]

Je ne sais donc pas quoi faire de plus, tout à l'air correctement configuré mais le sensor n'apparait pas :/

fmartinou, 2021-07-31:
Bonjour,

> Je viens de déployer la version 3.1.0 sur un raspi zero (car la 4.0.0 ne veut pas s'exécuter)

Il y a déjà une issue ouverte à ce sujet (https://github.com/fmartinou/teleinfo-mqtt/issues/9) ; je n'ai pas encore investigué.

> l'identifiant que j'ai défini dans le conteneur (linky) n'apparait pas dans le topic mqtt

Ce n'est pas normal en effet :/
Je vais regarder.

En attendant, tu peux supprimer la variable d'env HASS_IDENTIFIER pour lancer le conteneur avec les valeurs par défaut et ça fonctionnera.
(le HASS_IDENTIFIER n'a véritablement d'intérêt que pour ceux qui auraient plusieurs compteurs).

Exemple de ma configuration perso avec laquelle le sensor remonte bien sur home-assistant.

```yaml
  teleinfo:
      image: fmartinou/teleinfo-mqtt:4.0.0
      container_name: teleinfo
      restart: always
      devices:
          - /dev/ttyUSB0:/dev/ttyUSB0
      environment:
          - MQTT_URL=mqtt://mosquitto:1883
```

[image]

WLCD, 2021-07-31:
Bonjour @fmartinou, j'ai enlevé l'identifiant, sur vos conseils, je vais voir ce que ça donne.

fmartinou, 2021-08-05:
@WLCD ,
J'ai publié une nouvelle version **`5.0.0`**.
Le `identifier` n'existe plus.

Désormais le topic contient le ADCO (numéro de compteur).
Et si besoin, le nom du préfixe du topic peut être modifié par variable d'environnement (`MQTT_BASE_TOPIC`).

Tout est expliqué dans la doc mise à jour pour l'occasion :
https://fmartinou.github.io/teleinfo-mqtt

J'ai également rollbacké la version de node.js de 16 vers 14 car il se peut que ça soit ce changement-là qui fait planter l'appli sur les Pi Zéro.

Merci pour ton retour.

closed 2021-09-03 by fmartinou