fmartinou/teleinfo2mqtt #11: last_reset HA integration

ckd77, 2021-09-09:
Salut Fmartinou

Je pense que tu as vu, au niveau du mqtt last_reset va être obsolète ;) Du coup je me trouve avec une pollution dans les logs, et dans ton fichier hass.js tu utilise cette fonctionnalité.

<code>
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:213
Integration: Capteur (documentation, issues)
First occurred: 15:08:19 (2 occurrences)
Last logged: 15:08:19

Entity sensor.teleinfo_04**********_hchc_cost (<class 'homeassistant.components.energy.sensor.EnergyCostSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+energy%22
Entity sensor.teleinfo_04**********_hchp_cost (<class 'homeassistant.components.energy.sensor.EnergyCostSensor'>) with state_class measurement has set last_reset. Setting last_reset is deprecated and will be unsupported from Home Assistant Core 2021.11. Please update your configuration if state_class is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+energy%22</code>

Plus d'information sur https://www.home-assistant.io/blog/2021/09/01/release-20219/#release-202195---september-8 
section MQTT

fmartinou, 2021-09-09:
Merci pour l'info !

Faut dire que Home-Assistant se cherche encore avec cette histoire d'intégration à la plateforme Energie...
D'abord il fallait un `last_reset`, ensuite il fallait un `total_increasing`, maintenant il ne faut plus le `last_reset`... ;)

J'ai corrigé sur une branche.
Je teste et si c'est ok, je release le fix :)

fmartinou, 2021-09-09:
Fixed with version `6.0.2`.

closed 2021-09-09 by fmartinou

ckd77, 2021-09-09:
Nice job ;) 

En production la 6.0.2 ^^

```
CONTAINER ID   IMAGE                           COMMAND                  CREATED          STATUS          PORTS     NAMES
6fea2a4b6f6e   fmartinou/teleinfo-mqtt:6.0.2   "/usr/bin/entrypoint…"   11 minutes ago   Up 10 minutes             teleinfo2mqtt
```