fmartinou/teleinfo2mqtt #13: teleinfo-mqtt addons

nicoxygen, 2021-09-24:
Bonjour j'utilise votre projet depuis longtemps mais il y a peut je suis passé sous haos hébergé sur un raspberry pi et haos n'aime pas trop quand on rajoute un container manuellement. Penses-tu créer un addons du module ?

fmartinou, 2021-09-24:
Bonjour,

Je ne m'y suis pas intéressé car je n'utilise pas haos (je préfère garder la main sur tous les services).

C'est en partie pour ça que j'ai conçu ce projet en tant que service autonome qui est intégré via un standard (mqtt) plutôt qu'un composant intégré dans home-assistant.

Après je ne connais pas ce qu'il faut faire ni l'effort que ça représente pour en faire un addon! officiel.

As-tu des liens qui expliquent ce qu'il faudrait faire ?

nicoxygen, 2021-09-24:
Alors compliqué ce n'est pas, mais le temps de faire les tests, on pourrait y intégrer mqtt dans le container, c'est quelque chose que je maîtrise à peut près .. voir le module créé par mes soins 
https://www.haade.fr/blog/home-automation-smarthome-jeedom-homeassistant/tutos-haade-lab/frp-configurer-acces-distance-homeassistant/2/

Durant des années j'avais comme toi ha installer sur un container dans le NAS, mais avec le recul je me suis dit que si je vendais le logement il fallait que la domotique reste sur place sans monopoliser le NAS et divers container et puis haop a quelques subtilités que tu ne peux pas faire en virtuel ou en docker.

fmartinou, 2021-11-28:
Salut @nicoxygen ,

J'ai (enfin) créé l'addon...
L'URL du repository à utiliser dans Supervisor est : [https://github.com/fmartinou/hassio-addons.git](https://github.com/fmartinou/hassio-addons.git)

J'ai fait quelques tests dans une VM mais je n'ai pas fait un test de bout en bout car je n'utilise pas Hass.io moi-même.
Je pense que l'addon fonctionne ; j'ai juste une incertitude quand à savoir si le conteneur arrivera bien à accéder au `/dev/ttyUSBxxx` comme attendu pour pouvoir lire la teleinfo depuis le port série.

Si tu pouvais faire le beta-testeur, ça m'arrangerait bien... 😁 .

nicoxygen, 2021-11-28:
Top j’avais commencé à bosser dessus je fais ça et je debug s’il le faut

fmartinou, 2021-12-02:
Configuration pour l'add-on créé dans le repository `https://github.com/fmartinou/hassio-addons`.

1ère version de teleinfo-mqtt compatible = `6.1.0`

closed 2021-12-02 by fmartinou