fmartinou/teleinfo2mqtt #15: Add device class to iinst

matthew73210, 2021-12-29:
Hello,

In 'teleinfo-mqtt/app/mqtt/hass.js', you've only defined device classes to certain elements. Currently HA auto discovery is treating IINST as a base sensor and has no "unit of measurement". Thus graphs don't work, only history which looks weird. It is possible to use a template sensor to add a "unit of measurement", but this defeats the point of auto-discovery.

If possible, could you please add ' deviceClass = 'current'; ' to all current related tags?

I could do a PR but i'm away and won't be back till the end of Jan at the least.

This next part isn't related, I can open another issue, or PR:
I may have missed it but you don't specify if this add-on is for 'standard' or 'historical' mode. You probs know but they differ in multiple aspects. Glancing through your code i'm not sure it's compatible with the 'standard' mode, am I correct? If so this should probs be stated on the readme. 
Cheers

fmartinou, 2022-01-02:
Hi,

> could you please add ' deviceClass = 'current'; ' to all current related tags

Sure ! I just need to find some spare time for it 😁 .

> you don't specify if this add-on is for 'standard' or 'historical' mode

That's right; I would have to add this info to the documentation.
For now, **only the `historical` mode is supported**.

fmartinou, 2022-01-09:
Hi,

> could you please add ' deviceClass = 'current'; ' to all current related tags

>  Currently HA auto discovery is treating IINST as a base sensor and has no "unit of measurement"

That's strange ; it should already be implemented 🤔 
- [Class current](https://github.com/fmartinou/teleinfo-mqtt/blob/master/app/mqtt/hass.js#L28-L33)
- [Unit A ](https://github.com/fmartinou/teleinfo-mqtt/blob/master/app/mqtt/hass.js#L94-L99)

fmartinou, 2022-06-09:
I close the issue because of inactivity.
Feel free to reopen if the it is still relevant.

closed 2022-06-09 by fmartinou