fmartinou/teleinfo2mqtt #16: Add the possibility to upload production

Thyphus, 2022-01-13:
I have two linky, one for consumption and the other for the production.
Only the consumption has been took in account.

There are two TIC modes:
- HISTORIC with 1200 baudRate
- STANDARD with 9600 baudRate

My production linky is in STANDARD baudRate, so all frame are corrupted.

There are differents ID between consumption and production.

I can help you to improve the code to take in account the production if you want.

fmartinou, 2022-01-15:
Hi,

Thank you for your help!

Some people already asked for the support of the STANDARD mode.
I'd be glad to add support for it but unfortunately, I cannot access any Linky device with that mode enabled (neither a linky used for production).

If you can try to code id & test it on your end, it'd be perfect 😃 .
I can give you some guidance on where/how to make changes, etc.

fmartinou, 2022-01-15:
Basically, I think it would require...

Adding a new [`ticMode` configuration variable here](https://github.com/fmartinou/teleinfo-mqtt/blob/master/app/config/index.js#L7) with a `HISTORY` default value (for backward compatibility reason)

Adding the ability to override the default value [using a `TIC_MODE` env var here](https://github.com/fmartinou/teleinfo-mqtt/blob/master/app/config/index.js#L23) supporting `HISTORY` & `STANDARD` values.

Renaming [`app/teleinfo/index.js`](https://github.com/fmartinou/teleinfo-mqtt/blob/master/app/teleinfo/index.js) to `app/teleinfo/modeHistory.js`

Creating `app/teleinfo/modeStandard.js` to encapsulate all logic related to the standard mode.

Creating a new `app/teleinfo/index.js` script in charge of delegating to `app/teleinfo/modeHistory.js` or `app/teleinfo/modeStandard.js` according to the `ticMode` value.

Thyphus, 2022-01-15:
Hi,

It's fine if you can help me, I will test :). I don't know how to test modification. 

We need to create a branch no?

I agree for ticMode with the value HYSTORY and STANDARD.

I think we need to add another value like : type (PRODUCTION and CONSUMPTION) because the values are different (I will send you the output of linky). If I right understand, the standard mode come with more details than historic mode.
So we need to modify the section of code which manage HA data with new variable to push.

fmartinou, 2022-01-16:
Hi,

Thanks a million for your help 😃!

I think you can fork the repository, create a branch on your fork to start working on it, isn't it?
 
Coding & testing is very easy because it's just a regular node.js app.

Basic instructions:
1. Install node.js v14+
2. Clone the repository
3. Open a terminal on the `/app` directory
4. Install the project dependencies `npm ci`
5. Run locally the app with the appropiate env vars, for example `SERIAL=/dev/ttyUSB1 MQTT_URL=mqtt://192.1681.110:1883 npm start`

That's it!

Thyphus, 2022-01-20:
Hi,

I try different things.
I arrive to read the standard configuration but javascript is too slow... I miss maybe something in the code but the line I read is sometime cut or translated in another line. I protect the reading but sometimes there are some mistakes...

Can we talk about it with another canal please? Do you have a dev mail to share discord ID?

fmartinou, 2022-01-22:
Hi,

You should be able to find me on discord as `fmartinou#8623`.

Please don't be offensed if I'm not very available... I'm running out of time these days 😔.

I took a look at the changes you made on your fork; congrats!
Unfortunately, by just reading the code, for now, I don't understand why you get corrupted frames 🤔.

Mounicq, 2022-03-25:
feature/#18 branch actually generate Segmentation fault (core dumped).

fmartinou, 2022-04-02:
Hi @Mounicq ,

Can you provide the exact OS/Arch you're using?
- `linux/amd64`
- `linux/arm/v6`
- `linux/arm/v7`
- `linux/arm64`

And provide the logs of the core-dumped.

Thank you

cob94440, 2022-04-21:
Bonjour, du nouveau pour le mode standard ? Comment aider à part côté dev

tvenard, 2022-04-21:
> Bonjour, du nouveau pour le mode standard ? Comment aider à part côté dev
De mon coté j'ai plein de "Corrupted data" avec la branche feature/#18.
J'ai testé la branche production de Thyphus ce week-end avec mon compteur Linky en mode standard. Ça marche pas mal. 
[https://github.com/Thyphus/teleinfo-mqtt/tree/Production](url)

fmartinou, 2022-04-21:
Hello,

Je n'ai pas de dispo ces temps-ci pour finaliser les travaux.
La branche feature/#18 n'est pas utilisable en l'état (ni en mode historique ni en mode standard).

En attendant, vous pouvez utiliser la branche de @Thyphus mais vous devrez builder le projet et/ou l'image Docker vous-mêmes.

Je vais **essayer** de finaliser tout ça pendant ma semaine de vacances de pâques...

++

fmartinou, 2022-05-09:
La version `fmartinou/teleinfo-mqtt:feature__18` ne semble pas provoquer de régression par rapport à la version `6.1.0` en mode `HISTORY` ; c'est déjà ça.

Concernant le mode `STANDARD`, je viens de demander à mon fournisseur de basculer vers ce mode pour être en mesure de tester ; on verra si ça aboutit rapidement...

**En attendant, @Thyphus , @cob94440 , @tvenard , pouvez-vous essayer cette version en mode STANDARD et m'indiquer les erreurs que vous rencontrez ?**

Vous allez avoir dans les logs des erreurs `Corrupted data received...` ou `Corrupted line received...` ; est-ce que vous pouvez me les remonter ?
(ce sont sûrement des erreurs de code sur les contrôles du format des données pour certaines étiquettes).

Merci

cob94440, 2022-05-26:
Comment basculer l'addon dans cette version?
Tu as pu avoir ton changement de mode?
je m'y suis repris à deux fois avec edf car ils n'avaient pas rempli le bon formulaire. demande faite hier et à minuit c'etait effectif.

fmartinou, 2022-05-28:
Salut @cob94440 

> Tu as pu avoir ton changement de mode?
Toujours pas, non 😞 .
D'autant plus que je suis chez un fournisseur alternatif ; ça ne simplifie pas les choses...

> Comment basculer l'addon dans cette version?
Je pense que je vais créer une version `7.0.0` avec ces changements dans les prochains jours et la référencer côté addon HA.
Cela ne devrait pas poser de problème aux utilisateurs existants car le mode `HISTORY` semble fonctionner comme avant.

Concernant le mode `STANDARD`, je sais qu'il y a des erreurs dans le parsing des trames donc je compte sur vous pour me remonter ces erreurs et corriger en fonction 😉 .

cob94440, 2022-05-28:
En attendant je suis passé sur node red je peux te dire ce qui ressorts de
la lecture USB en 7 bits

fmartinou, 2022-06-06:
J'ai créé une version `7.0.0` qui intègre la possibilité de configurer le mode en `history` ou `standard`.
Cette version est référencée côté add-on HA (avec possibilité de choisir le mode depuis l'interface graphique de configuration de l'addon).

Je reste à l'écoute de vos retours pour corriger les problèmes de parsing des trames en mode standard...
++

fmartinou, 2022-06-06:
Les étiquettes implémentées sont les suivantes :
(je pense qu'elles y sont toutes 🤔 )

```
ADSC
CCAIN
CCAIN-1
CCASN
CCASN-1
DATE
DPM1
DPM2
DPM3
EAIT
EASD01
EASD02
EASD03
EASD04
EASF01
EASF02
EASF03
EASF04
EASF05
EASF06
EASF07
EASF08
EASF09
EASF10
EAST
ERQ1
ERQ2
ERQ3
ERQ4
FPM1
FPM2
FPM3
IRMS1
IRMS2
IRMS3
LTARF
MSG1
MSG2
NGTF
NJOURF
NJOURF+1
NTARF
PCOUP
PJOURF+1
PREF
PRM
RELAIS
SINSTI
SINSTS
SINSTS1
SINSTS2
SINSTS3
SMAXIN
SMAXIN-1
SMAXSN
SMAXSN-1
SMAXSN1
SMAXSN1-1
SMAXSN2
SMAXSN2-1
SMAXSN3
SMAXSN3-1
STGE
UMOY1
UMOY2
UMOY3
URMS1
URMS2
URMS3
VTIC
```

Mais il se peut que les règles de parsing de certaines d'entre-elles soient fausses...
A tester...

erwan33fr, 2022-06-08:
Bonjour,

Je viens de basculer en mode standard pour tester.
J'utilise teleinfo2mqtt version: 7.0.0 en addon sous home assistant.
En mode historique tout fonctionnait parfaitement :-)

En mode standard, les trames sont reçues (les valeurs reçu sont bonnes) mais il y a des caractères parasites à la fin => "Corrupted data received"

08:08:04.178  WARN teleinfo-mqtt: Corrupted data received [PREF	06	E
]
08:08:04.849  WARN teleinfo-mqtt: Corrupted data received [ADSC	02216XXXXXXX	5
]
08:08:04.867  WARN teleinfo-mqtt: Corrupted data received [VTIC	02	J
]
08:08:04.896  WARN teleinfo-mqtt: Corrupted data received [DATE	E220608080811		B
]
08:08:04.995  WARN teleinfo-mqtt: Corrupted data received [EAST	004552492	.
]
08:08:05.026  WARN teleinfo-mqtt: Corrupted data received [EASF01	002084472	=
]
08:08:05.059  WARN teleinfo-mqtt: Corrupted data received [EASF02	002468020	9
]
08:08:05.089  WARN teleinfo-mqtt: Corrupted data received [EASF03	000000000	$
]
08:08:05.122  WARN teleinfo-mqtt: Corrupted data received [EASF04	000000000	%
]
08:08:05.152  WARN teleinfo-mqtt: Corrupted data received [EASF05	000000000	&
]
08:08:05.187  WARN teleinfo-mqtt: Corrupted data received [EASF06	000000000	'
]
08:08:05.216  WARN teleinfo-mqtt: Corrupted data received [EASF07	000000000	(
]
08:08:05.296  WARN teleinfo-mqtt: Corrupted data received [EASF08	000000000	)
]
08:08:05.328  WARN teleinfo-mqtt: Corrupted data received [EASF09	000000000	*
]
08:08:05.362  WARN teleinfo-mqtt: Corrupted data received [EASF10	000000000	"
]
08:08:05.392  WARN teleinfo-mqtt: Corrupted data received [EASD01	004511642	7
]
08:08:05.424  WARN teleinfo-mqtt: Corrupted data received [EASD02	000040850	2
]
08:08:05.459  WARN teleinfo-mqtt: Corrupted data received [EASD03	000000000	"
]
08:08:05.474  WARN teleinfo-mqtt: Corrupted data received [EASD04	000000000	#
]
08:08:05.506  WARN teleinfo-mqtt: Corrupted data received [IRMS1	003	1
]
08:08:05.523  WARN teleinfo-mqtt: Corrupted data received [URMS1	228	F
]

erwan33fr, 2022-06-08:
Il y a chez Enedis des documents de référence qui donnent des informations détaillées sur les trames :

[http://www.enedis.fr/sites/default/files/Enedis-NOI-CPT_02E.pdf](url)

[http://www.enedis.fr/sites/default/files/Enedis-NOI-CPT_54E.pdf](url)

rdelville, 2022-06-08:
@erwan33fr 

Est-ce que ton compteur a été reconfiguré en mode standard ? Tu peux le vérifier dans le menu du compteur. (TIC -> historique ou standard)
Ton module teleinfo est-il compatible standard ?

Mon module est trop vieux pour le gérer. Je viens d'en recommander un et je pousserai mes résultats ici.

rdelville, 2022-06-08:
> En mode standard, les trames sont reçues (les valeurs reçu sont bonnes) mais il y a des caractères parasites à la fin => "Corrupted data received"

Au temps pour moi, je n'avais pas vu cette phrase

erwan33fr, 2022-06-08:
J'utilise un module Téléinfo USB au format rail DIN - Cartelectronic USBTIC :

[https://www.domotique-store.fr/domotique/modules-domotiques/detecteurs-capteurs-mesure/mesure-consommation-energetique/670-module-teleinfo-usb-au-format-rail-din-cartelectronic-usbtic.html](url)

fmartinou, 2022-06-08:
Bonjour @erwan33fr ,

Merci pour ton retour ; je vais pouvoir regarder de plus près.

Au feeling, je pense que le problème est là
https://github.com/fmartinou/teleinfo-mqtt/blob/master/app/teleinfo/standard/StandardTicMode.js#L50
...

J'investigue et je vous tiens au courant

erwan33fr, 2022-06-08:
Bonjour @fmartinou ,
Super :-)
Le mode standard est plus complet que le mode historique, donc j'aimerais pouvoir rester sur ce mode.
Si tu as besoin de logs supplémentaires, je suis dispo :-)

fmartinou, 2022-06-09:
@erwan33fr 

Can you give a try to the latest `7.0.1` version?
It should be working better... 🤞

erwan33fr, 2022-06-09:
Home Assistant ne me propose pas la mise à jour en  7.0.1 :-(

fmartinou, 2022-06-09:
Il faut peut être forcer la synchro des repositories HACS?

fmartinou, 2022-06-09:
Car j'ai bien déclaré la version :
https://github.com/fmartinou/hassio-addons/commit/8fd0b74283d26f83d71d092e2c229662d2690722

erwan33fr, 2022-06-09:
Oups, désolé, je n'avais pas forcé la mise à jour des addons

erwan33fr, 2022-06-09:
13:14:07.874  INFO teleinfo-mqtt:
    Starting teleinfo-mqtt with configuration = {
      emitInterval: 10,
      hassDiscovery: true,
      hassDiscoveryPrefix: 'homeassistant',
      logLevel: 'debug',
      mqttBaseTopic: 'teleinfo',
      mqttUrl: 'mqtt://core-mosquitto:1883',
      mqttUser: 'homeassistant',
      mqttPassword: '<hidden>',
      serial: '/dev/ttyUSB0',
      ticMode: 'standard'
    }
13:14:07.881  INFO teleinfo-mqtt: Connecting to port [/dev/ttyUSB0] with standard TIC mode
13:14:07.950  INFO teleinfo-mqtt: Connected to port [/dev/ttyUSB0]
13:14:07.951  INFO teleinfo-mqtt: Connecting to MQTT broker [mqtt://core-mosquitto:1883]
(node:7) UnhandledPromiseRejectionWarning: Error: Unsupported label [S]
    at StandardTicMode.getValue (/home/node/app/teleinfo/standard/StandardTicMode.js:126:19)
    at StandardTicMode.processData (/home/node/app/teleinfo/TicMode.js:92:28)
    at ReadLineParser.<anonymous> (/home/node/app/teleinfo/TicMode.js:70:46)
    at ReadLineParser.emit (events.js:400:28)
    at addChunk (internal/streams/readable.js:293:12)
    at readableAddChunk (internal/streams/readable.js:263:11)
    at ReadLineParser.Readable.push (internal/streams/readable.js:206:10)
    at ReadLineParser.Transform.push (internal/streams/transform.js:166:32)
    at ReadLineParser._transform (/home/node/app/node_modules/@serialport/parser-delimiter/lib/index.js:35:12)
    at ReadLineParser.Transform._read (internal/streams/transform.js:205:10)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
13:14:08.035  INFO teleinfo-mqtt: Connected to MQTT broker [mqtt://core-mosquitto:1883]

fmartinou, 2022-06-09:
Arf... je vois...
Le 1er bout de trame reçu au démarrage est incomplet et ne passe donc pas les validations...
=> Je fix

erwan33fr, 2022-06-09:
13:16:50.494  INFO teleinfo-mqtt:
    Starting teleinfo-mqtt with configuration = {
      emitInterval: 10,
      hassDiscovery: true,
      hassDiscoveryPrefix: 'homeassistant',
      logLevel: 'debug',
      mqttBaseTopic: 'teleinfo',
      mqttUrl: 'mqtt://core-mosquitto:1883',
      mqttUser: 'homeassistant',
      mqttPassword: '<hidden>',
      serial: '/dev/ttyUSB0',
      ticMode: 'standard'
    }
13:16:50.504  INFO teleinfo-mqtt: Connecting to port [/dev/ttyUSB0] with standard TIC mode
13:16:50.542  INFO teleinfo-mqtt: Connected to port [/dev/ttyUSB0]
13:16:50.544  INFO teleinfo-mqtt: Connecting to MQTT broker [mqtt://core-mosquitto:1883]
13:16:50.608  WARN teleinfo-mqtt: Corrupted data received [EASF06	000000000	'
]
13:16:50.614  WARN teleinfo-mqtt: Corrupted data received [EASF07	000000000	(
]
13:16:50.641  WARN teleinfo-mqtt: Corrupted data received [EASF08	000000000	)
]
13:16:50.673  WARN teleinfo-mqtt: Corrupted data received [EASF09	000000000	*
]
13:16:50.705  WARN teleinfo-mqtt: Corrupted data received [EASF10	000000000	"
]
13:16:50.737  WARN teleinfo-mqtt: Corrupted data received [EASD01	004532294	=
]
13:16:50.769  WARN teleinfo-mqtt: Corrupted data received [EASD02	000040850	2
]
13:16:50.801  WARN teleinfo-mqtt: Corrupted data received [EASD03	000000000	"
]
13:16:50.817  WARN teleinfo-mqtt: Corrupted data received [EASD04	000000000	#
]
13:16:50.849  WARN teleinfo-mqtt: Corrupted data received [IRMS1	002	0
]
13:16:50.865  WARN teleinfo-mqtt: Corrupted data received [URMS1	221	?
]
13:16:50.897  WARN teleinfo-mqtt: Corrupted data received [PREF	06	E
]
13:16:50.913  WARN teleinfo-mqtt: Corrupted data received [PCOUP	06	_
]
13:16:50.945  WARN teleinfo-mqtt: Corrupted data received [SINSTS	00400	J
]
13:16:50.977  WARN teleinfo-mqtt: Corrupted data received [SMAXSN	E220609060531	02740	9
]
13:16:51.022  INFO teleinfo-mqtt: Connected to MQTT broker [mqtt://core-mosquitto:1883]
13:16:51.025  WARN teleinfo-mqtt: Corrupted data received [SMAXSN-1	E220608071428	02840	^
]
13:16:51.059  WARN teleinfo-mqtt: Corrupted data received [CCASN	E220609130000	00406	9
]
13:16:51.105  WARN teleinfo-mqtt: Corrupted data received [CCASN-1	E220609123000	00352	Y
]
13:16:51.137  WARN teleinfo-mqtt: Corrupted data received [UMOY1	E220609131000	220	'
]
13:16:51.171  WARN teleinfo-mqtt: Corrupted data received [STGE	003A0401	>
]
13:16:51.249  WARN teleinfo-mqtt: Corrupted data received [PRM	16123733696527	>
]
13:16:51.281  WARN teleinfo-mqtt: Corrupted data received [RELAIS	000	B
]
13:16:51.297  WARN teleinfo-mqtt: Corrupted data received [NTARF	02	O
]
13:16:51.313  WARN teleinfo-mqtt: Corrupted data received [NJOURF	00	&
]
13:16:51.345  WARN teleinfo-mqtt: Corrupted data received [NJOURF+1	00	B
]
13:16:51.521  WARN teleinfo-mqtt: Corrupted data received [PJOURF+1	00008002 0200C001 08008002 0D1EC001 0F1E8002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	^
]
13:16:51.554  WARN teleinfo-mqtt: Corrupted data received [ADSC	022161XXXXXXX	5
]
13:16:51.569  WARN teleinfo-mqtt: Corrupted data received [VTIC	02	J
]
13:16:51.601  WARN teleinfo-mqtt: Corrupted data received [DATE	E220609131657		H
]
13:16:51.634  WARN teleinfo-mqtt: Corrupted data received [NGTF	H PLEINE/CREUSE 	\
]
13:16:51.714  WARN teleinfo-mqtt: Corrupted data received [EAST	004573144	+
]
13:16:51.747  WARN teleinfo-mqtt: Corrupted data received [EASF01	002093668	D
]
13:16:51.761  WARN teleinfo-mqtt: Corrupted data received [EASF02	002479476	J
]
13:16:51.793  WARN teleinfo-mqtt: Corrupted data received [EASF03	000000000	$
]
13:16:51.825  WARN teleinfo-mqtt: Corrupted data received [EASF04	000000000	%
]
13:16:51.859  WARN teleinfo-mqtt: Corrupted data received [EASF05	000000000	&
]
13:16:51.889  WARN teleinfo-mqtt: Corrupted data received [EASF06	000000000	'
]
13:16:51.921  WARN teleinfo-mqtt: Corrupted data received [EASF07	000000000	(
]
13:16:51.953  WARN teleinfo-mqtt: Corrupted data received [EASF08	000000000	)
]
13:16:51.985  WARN teleinfo-mqtt: Corrupted data received [EASF09	000000000	*
]
13:16:52.017  WARN teleinfo-mqtt: Corrupted data received [EASF10	000000000	"
]
13:16:52.049  WARN teleinfo-mqtt: Corrupted data received [EASD01	004532294	=
]
13:16:52.081  WARN teleinfo-mqtt: Corrupted data received [EASD02	000040850	2
]
13:16:52.113  WARN teleinfo-mqtt: Corrupted data received [EASD03	000000000	"
]

fmartinou, 2022-06-09:
Autant ton premier message, je l'explique et je vais corriger...
Autant ton second message avec les "WARN  Corrupted data received..." m'inquiètent plus car je ne comprends pas ce qui se passe 🤔

erwan33fr, 2022-06-09:
Les trames sont bonnes, les valeurs sont bonnes, mais il y a toujours un caractère ascii supplémentaire en fin de trame.

13:38:24.825  INFO teleinfo-mqtt:
    Starting teleinfo-mqtt with configuration = {
      emitInterval: 10,
      hassDiscovery: true,
      hassDiscoveryPrefix: 'homeassistant',
      logLevel: 'debug',
      mqttBaseTopic: 'teleinfo',
      mqttUrl: 'mqtt://core-mosquitto:1883',
      mqttUser: 'homeassistant',
      mqttPassword: '<hidden>',
      serial: '/dev/ttyUSB0',
      ticMode: 'standard'
    }
13:38:24.836  INFO teleinfo-mqtt: Connecting to port [/dev/ttyUSB0] with standard TIC mode
13:38:24.874  INFO teleinfo-mqtt: Connected to port [/dev/ttyUSB0]
13:38:24.876  INFO teleinfo-mqtt: Connecting to MQTT broker [mqtt://core-mosquitto:1883]
13:38:24.939  WARN teleinfo-mqtt: Corrupted data received [EASF06	000000000	'
]
13:38:24.945  WARN teleinfo-mqtt: Corrupted data received [EASF07	000000000	(
]
13:38:24.972  WARN teleinfo-mqtt: Corrupted data received [EASF08	000000000	)
]
13:38:25.002  WARN teleinfo-mqtt: Corrupted data received [EASF09	000000000	*
]
13:38:25.035  WARN teleinfo-mqtt: Corrupted data received [EASF10	000000000	"
]
13:38:25.066  WARN teleinfo-mqtt: Corrupted data received [EASD01	004532539	?
]
13:38:25.098  WARN teleinfo-mqtt: Corrupted data received [EASD02	000040850	2
]
13:38:25.130  WARN teleinfo-mqtt: Corrupted data received [EASD03	000000000	"
]
13:38:25.162  WARN teleinfo-mqtt: Corrupted data received [EASD04	000000000	#
]
13:38:25.178  WARN teleinfo-mqtt: Corrupted data received [IRMS1	005	3
]
13:38:25.210  WARN teleinfo-mqtt: Corrupted data received [URMS1	217	D
]
13:38:25.227  WARN teleinfo-mqtt: Corrupted data received [PREF	06	E
]
13:38:25.242  WARN teleinfo-mqtt: Corrupted data received [PCOUP	06	_
]
13:38:25.274  WARN teleinfo-mqtt: Corrupted data received [SINSTS	01121	K
]
13:38:25.306  WARN teleinfo-mqtt: Corrupted data received [SMAXSN	E220609060531	02740	9
]
13:38:25.311  INFO teleinfo-mqtt: Connected to MQTT broker [mqtt://core-mosquitto:1883]
13:38:25.354  WARN teleinfo-mqtt: Corrupted data received [SMAXSN-1	E220608071428	02840	^
]
13:38:25.402  WARN teleinfo-mqtt: Corrupted data received [CCASN	E220609133000	00382	?
]
13:38:25.434  WARN teleinfo-mqtt: Corrupted data received [CCASN-1	E220609130000	00406	W
]
13:38:25.514  WARN teleinfo-mqtt: Corrupted data received [UMOY1	E220609133000	217	/
]
13:38:25.546  WARN teleinfo-mqtt: Corrupted data received [STGE	003A0000	9
]
13:38:25.626  WARN teleinfo-mqtt: Corrupted data received [PRM	16123733696527	>
]
13:38:25.658  WARN teleinfo-mqtt: Corrupted data received [RELAIS	001	C
]
13:38:25.674  WARN teleinfo-mqtt: Corrupted data received [NTARF	01	N
]
13:38:25.690  WARN teleinfo-mqtt: Corrupted data received [NJOURF	00	&
]
13:38:25.722  WARN teleinfo-mqtt: Corrupted data received [NJOURF+1	00	B
]
13:38:25.851  WARN teleinfo-mqtt: Corrupted data received [PJOURF+1	00008002 0200C001 08008002 0D1EC001 0F1E8002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	^
]
13:38:25.884  WARN teleinfo-mqtt: Corrupted data received [ADSC	02216XXXXXX	5
]
13:38:25.898  WARN teleinfo-mqtt: Corrupted data received [VTIC	02	J
]
13:38:25.931  WARN teleinfo-mqtt: Corrupted data received [DATE	E220609133832		E
]
13:38:25.980  WARN teleinfo-mqtt: Corrupted data received [NGTF	H PLEINE/CREUSE 	\
]
13:38:26.042  WARN teleinfo-mqtt: Corrupted data received [EAST	004573389	6
]
13:38:26.074  WARN teleinfo-mqtt: Corrupted data received [EASF01	002093825	?
]
13:38:26.107  WARN teleinfo-mqtt: Corrupted data received [EASF02	002479564	H
]
13:38:26.138  WARN teleinfo-mqtt: Corrupted data received [EASF03	000000000	$
]
13:38:26.170  WARN teleinfo-mqtt: Corrupted data received [EASF04	000000000	%
]
13:38:26.202  WARN teleinfo-mqtt: Corrupted data received [EASF05	000000000	&
]
13:38:26.233  WARN teleinfo-mqtt: Corrupted data received [EASF06	000000000	'
]
13:38:26.266  WARN teleinfo-mqtt: Corrupted data received [EASF07	000000000	(
]
13:38:26.298  WARN teleinfo-mqtt: Corrupted data received [EASF08	000000000	)
]
13:38:26.314  WARN teleinfo-mqtt: Corrupted data received [EASF09	000000000	*
]
13:38:26.345  WARN teleinfo-mqtt: Corrupted data received [EASF10	000000000	"
]
13:38:26.378  WARN teleinfo-mqtt: Corrupted data received [EASD01	004532539	?
]
13:38:26.410  WARN teleinfo-mqtt: Corrupted data received [EASD02	000040850	2
]
13:38:26.442  WARN teleinfo-mqtt: Corrupted data received [EASD03	000000000	"
]
13:38:26.522  WARN teleinfo-mqtt: Corrupted data received [EASD04	000000000	#
]
13:38:26.554  WARN teleinfo-mqtt: Corrupted data received [IRMS1	005	3
]
13:38:26.570  WARN teleinfo-mqtt: Corrupted data received [URMS1	217	D
]
13:38:26.586  WARN teleinfo-mqtt: Corrupted data received [PREF	06	E
]
13:38:26.618  WARN teleinfo-mqtt: Corrupted data received [PCOUP	06	_
]
13:38:26.633  WARN teleinfo-mqtt: Corrupted data received [SINSTS	01126	P
]
13:38:26.682  WARN teleinfo-mqtt: Corrupted data received [SMAXSN	E220609060531	02740	9
]
13:38:26.714  WARN teleinfo-mqtt: Corrupted data received [SMAXSN-1	E220608071428	02840	^
]

fmartinou, 2022-06-09:
Le dernier caractère, c'est le checksum ; c'est normal qu'il soit là.
Ce n'est pas ça qui doit poser problème.
Ca doit être autre chose...

fmartinou, 2022-06-09:
Par contre, il y a peut-être un caractère de retour charriot à la fin de la trame.
Ce qui expliquerait que tes logs soient sur 2 lignes (sauf si c'est ton copier-coller qui a provoqué ça ?)

```
13:38:24.939 WARN teleinfo-mqtt: Corrupted data received [EASF06 000000000 '
] <-- Ici
```
Ca expliquerait le problème en tout cas 🤔

erwan33fr, 2022-06-09:
[image]

erwan33fr, 2022-06-09:
👍🤞

fmartinou, 2022-06-09:
@erwan33fr 
Peux-tu tester la version `7.0.2`, stp ?
Et si encore des problème, peux-tu activer les traces en niveau debug sur l'addon (`LOG_LEVEL=debug`) et me remonter les logs ?

Merci :)
(ça finira bien par marcher...)

erwan33fr, 2022-06-09:
👍 ça fonctionne 👏👏👏
Je posterai les traces et des captures d'écrans dans la soirée.
Il me reste à supprimer de l'appareil les anciens capteurs qui avaient été créés avec le mode historique et qui ne sont plus utilisés en mode standard.
Les nouveaux capteurs sont bien présents et bien renseignés 👍

closed 2022-06-09 by fmartinou

erwan33fr, 2022-06-09:
21:46:04.141 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.173 DEBUG teleinfo-mqtt: Raw frame [EASD04	000000000	#
]
21:46:04.173 DEBUG teleinfo-mqtt: Split frame [EASD04,000000000,#
]
21:46:04.173 DEBUG teleinfo-mqtt: Value for label EASD04 = 000000000
21:46:04.173 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.205 DEBUG teleinfo-mqtt: Raw frame [IRMS1	003	1
]
21:46:04.205 DEBUG teleinfo-mqtt: Split frame [IRMS1,003,1
]
21:46:04.205 DEBUG teleinfo-mqtt: Value for label IRMS1 = 003
21:46:04.205 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.221 DEBUG teleinfo-mqtt: Raw frame [URMS1	229	G
]
21:46:04.221 DEBUG teleinfo-mqtt: Split frame [URMS1,229,G
]
21:46:04.221 DEBUG teleinfo-mqtt: Value for label URMS1 = 229
21:46:04.221 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.239 DEBUG teleinfo-mqtt: Raw frame [PREF	06	E
]
21:46:04.239 DEBUG teleinfo-mqtt: Split frame [PREF,06,E
]
21:46:04.239 DEBUG teleinfo-mqtt: Value for label PREF = 06
21:46:04.239 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.253 DEBUG teleinfo-mqtt: Raw frame [PCOUP	06	_
]
21:46:04.253 DEBUG teleinfo-mqtt: Split frame [PCOUP,06,_
]
21:46:04.253 DEBUG teleinfo-mqtt: Value for label PCOUP = 06
21:46:04.253 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.285 DEBUG teleinfo-mqtt: Raw frame [SINSTS	00651	R
]
21:46:04.285 DEBUG teleinfo-mqtt: Split frame [SINSTS,00651,R
]
21:46:04.285 DEBUG teleinfo-mqtt: Value for label SINSTS = 00651
21:46:04.285 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.335 DEBUG teleinfo-mqtt: Raw frame [SMAXSN	E220609060531	02740	9
]
21:46:04.335 DEBUG teleinfo-mqtt: Split frame [SMAXSN,E220609060531,02740,9
]
21:46:04.335 DEBUG teleinfo-mqtt: Value for label SMAXSN = 02740
21:46:04.335 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.365 DEBUG teleinfo-mqtt: Raw frame [SMAXSN-1	E220608071428	02840	^
]
21:46:04.365 DEBUG teleinfo-mqtt: Split frame [SMAXSN-1,E220608071428,02840,^
]
21:46:04.365 DEBUG teleinfo-mqtt: Value for label SMAXSN-1 = 02840
21:46:04.365 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.414 DEBUG teleinfo-mqtt: Raw frame [CCASN	E220609213000	00496	D
]
21:46:04.415 DEBUG teleinfo-mqtt: Split frame [CCASN,E220609213000,00496,D
]
21:46:04.415 DEBUG teleinfo-mqtt: Value for label CCASN = 00496
21:46:04.415 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.446 DEBUG teleinfo-mqtt: Raw frame [CCASN-1	E220609210000	00886	"
]
21:46:04.446 DEBUG teleinfo-mqtt: Split frame [CCASN-1,E220609210000,00886,"
]
21:46:04.446 DEBUG teleinfo-mqtt: Value for label CCASN-1 = 00886
21:46:04.446 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.495 DEBUG teleinfo-mqtt: Raw frame [UMOY1	E220609214000	228	1
]
21:46:04.495 DEBUG teleinfo-mqtt: Split frame [UMOY1,E220609214000,228,1
]
21:46:04.495 DEBUG teleinfo-mqtt: Value for label UMOY1 = 228
21:46:04.495 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:04.509 DEBUG teleinfo-mqtt: Raw frame [STGE	003A0401	>
]
21:46:04.509 DEBUG teleinfo-mqtt: Split frame [STGE,003A0401,>
[... 109 lines cut ...]
21:46:05.213 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:05.245 DEBUG teleinfo-mqtt: Raw frame [EASF06	000000000	'
]
21:46:05.245 DEBUG teleinfo-mqtt: Split frame [EASF06,000000000,'
]
21:46:05.245 DEBUG teleinfo-mqtt: Value for label EASF06 = 000000000
21:46:05.245 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:05.278 DEBUG teleinfo-mqtt: Raw frame [EASF07	000000000	(
]
21:46:05.278 DEBUG teleinfo-mqtt: Split frame [EASF07,000000000,(
]
21:46:05.278 DEBUG teleinfo-mqtt: Value for label EASF07 = 000000000
21:46:05.279 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:05.309 DEBUG teleinfo-mqtt: Raw frame [EASF08	000000000	)
]
21:46:05.310 DEBUG teleinfo-mqtt: Split frame [EASF08,000000000,)
]
21:46:05.310 DEBUG teleinfo-mqtt: Value for label EASF08 = 000000000
21:46:05.310 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]
21:46:05.341 DEBUG teleinfo-mqtt: Raw frame [EASF09	000000000	*
]
21:46:05.341 DEBUG teleinfo-mqtt: Split frame [EASF09,000000000,*
]
21:46:05.342 DEBUG teleinfo-mqtt: Value for label EASF09 = 000000000
21:46:05.342 DEBUG teleinfo-mqtt: Frame parsed [[object Object]]

erwan33fr, 2022-09-16:
Il semble qu'il y a un bug avec le mode standard.
à chaque émission de messages MQTT, dans les logs de Home Assistant, il y a : 

```
Détails du journal (ERREUR)
Logger: homeassistant.helpers.template
Source: helpers/template.py:582
First occurred: 17:05:57 (80 occurrences)
Last logged: 17:07:09

Template variable error: 'dict object' has no attribute 'PTEC' when rendering '{{ value_json.PTEC.value }}'
Template variable error: 'dict object' has no attribute 'IINST' when rendering '{{ value_json.IINST.value }}'
Template variable error: 'dict object' has no attribute 'IMAX' when rendering '{{ value_json.IMAX.value }}'
Template variable error: 'dict object' has no attribute 'PAPP' when rendering '{{ value_json.PAPP.value }}'
Template variable error: 'dict object' has no attribute 'HHPHC' when rendering '{{ value_json.HHPHC.value }}'
```

erwan33fr, 2022-09-16:
J'ai l'impression qu'il s'agit de variable qui sont resté de ma précédente configuration en mode "historique".
Et lorsque je suis passé en mode standard, des variables du mode "historique" semblent encore présente.

erwan33fr, 2022-10-16:
Mauvaise nouvelle, depuis la dernière mise à jour Home Assistant, teleinfo2mqtt ne fonctionne plus :-(

Trace de Mosquitto broker :
2022-10-16 20:30:27: New connection from 172.30.33.3:51920 on port 1883.
2022-10-16 20:30:27: New client connected from 172.30.33.3:51920 as mqttjs_a57f155e (p2, c1, k60, u'homeassistant').
2022-10-16 20:30:37: Client mqttjs_a57f155e disconnected due to protocol error.

Trace de teleinfo2mqtt : 
20:31:59.289  INFO teleinfo2mqtt: Connecting to port [/dev/ttyUSB0] with standard TIC mode
20:31:59.344  INFO teleinfo2mqtt: Connected to port [/dev/ttyUSB0]
20:31:59.346  INFO teleinfo2mqtt: Connecting to MQTT broker [mqtt://core-mosquitto:1883]
20:31:59.409  INFO teleinfo2mqtt: Connected to MQTT broker [mqtt://core-mosquitto:1883]
.
.
.
20:32:10.028  WARN teleinfo2mqtt: Error when publishing to the mqtt broker (read ECONNRESET)
20:32:11.028  INFO teleinfo2mqtt: Reconnecting to the MQTT broker...

:-(

fmartinou, 2022-10-16:
Je suis en dernière version de homeassistant (2022.10.4) et de teleinfo2mqtt (8.1.2) et je n'ai pas de problème 🤔

erwan33fr, 2022-10-16:
Jusqu'à la mise à jour, ça fonctionnait :-(
Comment faite pour passer à la version précédente pour vérifier ?

erwan33fr, 2022-10-16:
Ma configuration : 

EMIT_INTERVAL: 10
HASS_DISCOVERY_PREFIX: homeassistant
LOG_LEVEL: info
MQTT_BASE_TOPIC: teleinfo
MQTT_URL: mqtt://core-mosquitto:1883
MQTT_USER: homeassistant
MQTT_PASSWORD: *************************
SERIAL: /dev/ttyUSB0
TIC_MODE: standard

fmartinou, 2022-10-16:
Vous avez une version à jour de mosquito?

erwan33fr, 2022-10-16:
Version 6.1.3
(Module home assistant)

rdelville, 2022-10-17:
Bonjour,

Pour tester, j'ai mis à jour ma version de teleinfo2mqtt (8.0.2 -> 8.1.2) avec la dernière version de HA (2022.10.4) et mosquito (6.1.3  module HA). Je n'ai plus aucune remontée.

J'ai fait un retour arrière à la version 8.0.2 et les données sont revenues. Si j'ai un peu de temps, je retesterai ce soir.

fmartinou, 2022-10-17:
Bonjour,

Entre la `8.0.2` et la `8.1.2`, il y a plusieurs versions intermédiaires (`8.1.0` et `8.1.1`).
Savoir à partir de quelle version exactement, ça ne fonctionne plus m'aidera à identifier le changement concerné.

erwan33fr, 2022-10-17:
> Bonjour,
>
> Pour tester, j'ai mis à jour ma version de teleinfo2mqtt (8.0.2 -> 8.1.2) avec la dernière version de HA (2022.10.4) et mosquito (6.1.3 module HA). Je n'ai plus aucune remontée.
>
> J'ai fait [...]

Comment faire un retour arrière ?

rdelville, 2022-10-17:
> Comment faire un retour arrière ?

Il suffit de préciser le tag que tu veux lorsque tu lances ton docker. Exemple  : 

```
docker run -d --name teleinfo2mqtt \
  --device=/dev/ttyUSB0:/dev/ttyUSB0 \
  -e MQTT_URL=mqtt://my_mqtt_broker:1883 \
  fmartinou/teleinfo2mqt:8.1.1
```

Je confirme également qu'en 8.1.1, ça fonctionne.

erwan33fr, 2022-10-17:
Merci pour le retour arrière, mais j'utilise un module Home Assistant.
Pas de docker.

fmartinou, 2022-10-18:
J'ai buildé une version en downgradant la lib `async-mqtt`
=> `fmartinou/teleinfo2mqtt:fix_downgrade_async_mqtt`

Si qqn pour qui la dernière version ne fonctionne pas pouvait la tester ; ça permettrait de confirmer que c'est bien la cause de la régression.

rdelville, 2022-10-18:
> J'ai buildé une version en downgradant la lib `async-mqtt` => `fmartinou/teleinfo2mqtt:fix_downgrade_async_mqtt`
> 
> Si qqn pour qui la dernière version ne fonctionne pas pouvait la tester ; ça permettrait de confirmer que c'est bien la cause de la régression.

Bonsoir @fmartinou,

Aucun changement avec cette version.Je suis repassé en 8.1.1 pour avoir des remontés.

Pendant le test, j'ai passé le log_level à debug. Aucune erreur remontée. Les trames sont bien parsées et lors du publish, je vois bien les valeurs actualisées. Sauf que sur HA, rien ne change.

Je peux réaliser d'autres tests au besoin.

fmartinou, 2022-10-18:
Alors là, on est mal car la seule diff entre les 2, c'est ça :

https://github.com/fmartinou/teleinfo2mqtt/compare/8.1.1...fix/downgrade_async_mqtt#diff-331b184a1d6d99b37bf119166d104f0c0d5fd2bee3201d0e2900082bc181c006R134

rdelville, 2022-10-18:
Je viens de consulter les logs de Mosquito broker, j'ai plusieurs erreurs au niveau du brooker avec l'image teleinfo2mqtt:fix_downgrade_async_mqtt

> 2022-10-18 19:23:35: New connection from 172.30.32.1:42364 on port 1883.
> 2022-10-18 19:23:35: New client connected from 172.30.32.1:42364 as mqttjs_1c632ded (p2, c1, k60, u'mqtt_user').
> 2022-10-18 19:23:44: Client mqttjs_1c632ded disconnected due to protocol error.

Lorsque je suis en version 8.1.1, je n'ai aucune erreur.

fmartinou, 2022-10-18:
Outre le code, c'est peut-être la version de nodejs qui a changé entre le moment où la version qui marche a été buildée et le moment où la version qui ne marche pas a été buildée.
Je vais vérifier ce point.

fmartinou, 2022-10-18:
```
$ docker run --rm fmartinou/teleinfo2mqtt:8.1.1 node -v
v14.20.0

$ docker run --rm fmartinou/teleinfo2mqtt:8.1.2 node -v
v14.20.1

$ docker run --rm fmartinou/teleinfo2mqtt:fix_downgrade_async_mqtt node -v
v14.20.1
```

[17 more replies]