fmartinou/teleinfo2mqtt #17: Module teleinfo usb

cob94440, 2022-02-16:
Bonjour, 
Cet Addon est il compatible avec la clé USB suivante:
https://www.domotique-store.fr/domotique/modules-domotiques/detecteurs-capteurs-mesure/mesure-consommation-energetique/672-module-teleinfo-usb-cartelectronic-usbticlc-win.html

Merci d'avance de votre retour.
Chris

fmartinou, 2022-02-16:
Bonjour,

Cette app fonctionne avec n'importe quel matériel qui met à disposition les trames issues de la teleinfo sur un port série.

En suivant ton lien et en parcourant les questions/réponses des utilisateurs, le marchand détaille ceci

> Le USBTIC est en fait un simple "MODEM", il ne fait qu'adapter le signal téléinfo (bus série propriétaire) vers de l'USB mais en transmettant les trames telles que le compteur les transmet.
En gros, vous pouvez parfaitement l'utiliser sur un raspberry avec Linux mais cela nécessitera de développer un script sur mesure (ou adapter un script télé-information existant) afin de traiter les trames brutes reçues du compteur

Donc d'après ces informations ; cela fonctionnera.

NB
Perso j'utilise le module suivent et cela fonctionne parfaitement:
https://www.domotique-store.fr/domotique/modules-domotiques/detecteurs-capteurs-mesure/mesure-consommation-energetique/425-gce-electronics-interface-teleinformation-usb-pour-compteur-electrique.html

Cdt

cob94440, 2022-02-19:
Mon but c'est remplacer suivi conso sur jeedom. Mon dongle est sur un raspi avec jeedom. 
Un plugin permet il de mettre les couts sur HA?

fmartinou, 2022-02-19:
Bonjour,

Les données de teleinfo remontées par `teleinfo-to-mqtt` peuvent être utilisées pour configurer [le module `Energy`](https://www.home-assistant.io/docs/energy/) de HA.
Et donc de suivre les coûts [comme indiqué dans le documentation](https://fmartinou.github.io/teleinfo-mqtt/#/hass/?id=home-assistant-energy).
[image]

closed 2022-03-05 by fmartinou

cob94440, 2022-04-04:
Le module fonctionnent correctement. 
Il me reste à voir si mon souci vient du mode historique.

fmartinou, 2022-04-04:
> Il me reste à voir si mon souci vient du mode historique.

Pour information, le mode HISTORIQUE est le seul mode supporté pour le moment (le support du mode STANDARD est en cours d'implémentation).