fmartinou/teleinfo2mqtt #20: EDF Zen Plus

cob94440, 2022-03-25:
Hello I have EDF zen plus and the addon detect base tarif. Not HP or HC.
I Have wednesday, saturday and sunday where HP are at HC cost.
And rest of time 12h30-15h30 HC and 03h00-08h00 HC.
What is wrong?

cob94440, 2022-03-25:
[image]

cob94440, 2022-04-11:
Tu as une idée de pourquoi le tarif est en TH? Toutes heures.

fmartinou, 2022-04-11:
Les propriétés remontées via mqtt sont directement les étiquettes trouvées dans les trames reçues.

Donc j'imagine que la trame reçue a pour valeur BASE pour l'étiquette OPTARIF.

Je vais regarder la spécification teleinfo en mode historique mais il se peut que ton abonnement zen plus ne soit pas supporté par la spec.

fmartinou, 2022-04-11:
La spec

https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.enedis.fr/media/2035/download&ved=2ahUKEwj-5cSE24z3AhXCyYUKHbcdCf8QFnoECAMQAQ&usg=AOvVaw0Bn9Pv64VK2dhbtmrZD8YD

fmartinou, 2022-04-11:
Ça semble être les infos remontées par le compteur qui sont en cause.

https://forum.gce-electronics.com/t/eco-device-avec-abonnement-mes-jours-zen/13764/2

cob94440, 2022-04-11:
Visiblement le dernier message indique que la personne est passé en standard

fmartinou, 2022-04-11:
Je ne connais pas la procédure pour passer le compteur en mode standard ; enedis doit-il intervenir ?

teleinfo2mqtt n'est pas encore compatible avec le mode standard mais c'est en cours.

cob94440, 2022-04-11:
C'est un simple appel à mon fournisseur et ils font la demande a enedis.
J'ai vu sur jeedom que ça remonte davantage d'informations.
Sur mon linky je vois bien les différentes périodes HP/hc. A savoir
qu'après la complexité c'est que le mercredi et le weekend ça reste des
HP/hc mais les HP sont au tarif hc. Et le reste de la semaine c'est le
classique HP/hc

cob94440, 2022-04-14:
De ce que j'ai comme info les différences sont le nom du mode et la vitesse qui est a 9600 bauds

closed 2022-05-09 by fmartinou