fmartinou/teleinfo2mqtt #26: Docker image 7.0.0 casse le mode historique sur Raspberry

rtorrente, 2022-06-08:
Je remonte un bug qui semble être apparu sur la version 7.0.0 publiée récemment. 

Après upgrade du container docker sous la dernière latest (sha256:7627dd25d7dac7214b3add0479b42d0a1b6f26fa51614e533a47c6b5270ccb15) et le container est en bootloop sur une erreur de segmentation

```
22:11:19.787  INFO teleinfo-mqtt:
    Starting teleinfo-mqtt with configuration = {
      emitInterval: '5',
      hassDiscovery: true,
      hassDiscoveryPrefix: 'homeassistant',
      logLevel: 'info',
      mqttBaseTopic: 'teleinfo',
      mqttUrl: 'mqtt://mqtt',
      mqttUser: undefined,
      mqttPassword: '<hidden>',
      serial: '/dev/ttyAMA0',
      ticMode: 'history'
    }
22:11:19.813  INFO teleinfo-mqtt: Connecting to port [/dev/ttyAMA0] with history TIC mode
Segmentation fault (core dumped)
```

En repassant sur l'image de tag 6.1.0, tout fonctionne immédiatement

fmartinou, 2022-06-08:
It's likely an issue with the node-serialport package in it's latest version (10+).

https://github.com/serialport/node-serialport/issues/2438

I'm gonna take a look and downgrade it if so.

fmartinou, 2022-06-09:
@rtorrente @Bobdugrand8

I've downgraded the serialport from `v10` to `v9` as the others also did (zigbee2mqtt, zwave2mqtt, node-red...).
Can you give a try to the latest `7.0.1`?
Thanks

closed 2022-06-09 by rtorrente

ripp2003, 2022-06-09:
@martinou : Thanks for the fix, it's ok