fmartinou/teleinfo2mqtt #29: Warnings en mode standard

rdelville, 2022-06-15:
Bonjour,

Je viens de passer en mode standard sur mon compteur Linky. J'ai des warnings qui apparaissent dans les logs en mode info :+1: 

```
09:15:34.489  WARN teleinfo2mqtt: Invalid value received for label PJOURF+1 [00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE]
09:15:35.401  WARN teleinfo2mqtt: Invalid value received for label PJOURF+1 [00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE]
09:15:36.329  WARN teleinfo2mqtt: Invalid value received for label PJOURF+1 [00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE]
09:15:37.258  WARN teleinfo2mqtt: Invalid value received for label PJOURF+1 [00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE]
```

### Infos
- Version utilisée : teleinfo2mqtt:8.0.0
- Module teleinfo compatible standard (https://www.cartelectronic.fr/teleinfo-compteur-enedis/127-teleinfo-1-compteur-usb-lc.html)
- Mode TIC : standard

### Test en ligne de commande
```
stty -F /dev/ttyAMA0 9600 sane evenp parenb cs7 -crtscts
cat /dev/ttyUSB0
```
```
ADSC	0XXXXXXXXXX	J
VTIC	02	J
DATE	E220615092559		L
NGTF	H PLEINE/CREUSE 	\
LTARF	INDEX INACTIF 2 	3
EAST	018378744	9
EASF01	009094481	E
EASF02	009284263	E
EASF03	000000000	$
EASF04	000000000	%
EASF05	000000000	&
EASF06	000000000	'
EASF07	000000000	(
EASF08	000000000	)
EASF09	000000000	*
EASF10	000000000	"
EASD01	007311917	=
EASD02	007471779	K
EASD03	001782564	C
EASD04	001812484	?
IRMS1	001	/
URMS1	233	B
PREF	06	E
PCOUP	06	_
SINSTS	00348	U
SMAXSN	E220615075131	01835	<
SMAXSN-1	E220614190610	03177	Z
CCASN	E220615090000	00262	;
CCASN-1	E220615083000	00318	]
UMOY1	E220615092000	233	.
STGE	003A4401	B
MSG1	PAS DE          MESSAGE         	<
PRM	1XXXXXXXXXX	J
RELAIS	000	B
NTARF	02	O
NJOURF	00	&
NJOURF+1	00	B
PJOURF+1	00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	Z
```

### mqtt2teleinfo en mode debug
```]9:13:50.051 DEBUG teleinfo2mqtt: Raw frame [ADSC	0XXXXXXXXXX	J
]9:13:50.052 DEBUG teleinfo2mqtt: Split frame [ADSC,0XXXXXXXXXX,J
09:13:50.052 DEBUG teleinfo2mqtt: Value for label ADSC = 0XXXXXXXXXX
09:13:50.052 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
09:13:50.055 DEBUG teleinfo2mqtt: Ignoring MQTT emission because of emit interval (Emit interval : 10 - Last emit time : 1655277228198 - Current time : 1655277230055
]9:13:50.056 DEBUG teleinfo2mqtt: Raw frame [VTIC	02	J
]9:13:50.058 DEBUG teleinfo2mqtt: Split frame [VTIC,02,J
[... 126 lines cut ...]
]9:13:50.772 DEBUG teleinfo2mqtt: Raw frame [RELAIS	000	B
]9:13:50.772 DEBUG teleinfo2mqtt: Split frame [RELAIS,000,B
09:13:50.772 DEBUG teleinfo2mqtt: Value for label RELAIS = 000
09:13:50.772 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
]9:13:50.787 DEBUG teleinfo2mqtt: Raw frame [NTARF	02	O
]9:13:50.788 DEBUG teleinfo2mqtt: Split frame [NTARF,02,O
09:13:50.788 DEBUG teleinfo2mqtt: Value for label NTARF = 02
09:13:50.788 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
]9:13:50.788 DEBUG teleinfo2mqtt: Raw frame [NJOURF	00	&
]9:13:50.789 DEBUG teleinfo2mqtt: Split frame [NJOURF,00,&
09:13:50.789 DEBUG teleinfo2mqtt: Value for label NJOURF = 00
09:13:50.789 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
]9:13:50.803 DEBUG teleinfo2mqtt: Raw frame [NJOURF+1	00	B
]9:13:50.804 DEBUG teleinfo2mqtt: Split frame [NJOURF+1,00,B
09:13:50.804 DEBUG teleinfo2mqtt: Value for label NJOURF+1 = 00
09:13:50.804 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
]9:13:50.948 DEBUG teleinfo2mqtt: Raw frame [PJOURF+1	00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	Z
]9:13:50.949 DEBUG teleinfo2mqtt: Split frame [PJOURF+1,00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE,Z
09:13:50.949  WARN teleinfo2mqtt: Invalid value received for label PJOURF+1 [00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE]
```
Le champ `DATE` récupère le checksum et non la valeur. Pour `PJOURF+1`, je ne vois pas le souci.

Romain

PS: J"ai anonymisé quelques valeurs.

rdelville, 2022-06-15:
Dans le log de debug, on peut voir le traitement sur le champ `DATE`

```
]9:13:50.084 DEBUG teleinfo2mqtt: Raw frame [DATE	E220615091349		H
]9:13:50.084 DEBUG teleinfo2mqtt: Split frame [DATE,E220615091349,H
09:13:50.084 DEBUG teleinfo2mqtt: Value for label DATE = H
```
`DATE` prend la valeur H au lieu de E220615091349

fmartinou, 2022-06-15:
Ah ok je vois.
C'est une erreur du même type que l'autre.
=> Je vais fixer

fmartinou, 2022-06-15:
Les 2 points devraient être résolus dans la dernière version `8.0.1`.
Concernant le label `DATE`, la `value` sera vide ; le contenu sera dans la propriété `timestamp`.

Exemple :
```json
{
   "DATE":{
      "raw":"",
      "value":"",
      "timestamp":{
         "date":"2022-06-09T06:05:31.000Z",
         "dst":"summer"
      }
   }
}
```

C'est pour être cohérent par rapport au reste des données puisque cette étiquette est définie par ENEDIS comme ayant une valeur nulle et un horodatage :
[image]

rdelville, 2022-06-15:
Le label `DATE` est bien à vide.

```
1:01:34.510 DEBUG teleinfo2mqtt: Raw frame [DATE	E220615210133		8
]1:01:34.510 DEBUG teleinfo2mqtt: Split frame [DATE,E220615210133,8
21:01:34.511 DEBUG teleinfo2mqtt: Value for label DATE = 
```

Plus de warning pour le label `PJOURF+1`
```
]1:01:35.374 DEBUG teleinfo2mqtt: Raw frame [PJOURF+1	00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	Z
]1:01:35.375 DEBUG teleinfo2mqtt: Split frame [PJOURF+1,00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE,Z
21:01:35.375 DEBUG teleinfo2mqtt: Value for label PJOURF+1 = 00008002 0050C001 06508002 1220C001 14208002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE
```

Merci pour le fix. :)

Romain

closed 2022-06-15 by fmartinou