fmartinou/teleinfo2mqtt #34: problème serial suite passage en 8.1.0

fredarro, 2022-09-25:
```
19:31:09.625 ERROR teleinfo2mqtt: Error: No such file or directory, cannot open /dev/ttyUSB0
    Error: Error: No such file or directory, cannot open /dev/ttyUSB0
```

alors que le stack est le suivant:
 ```version: "3.8"
services:
 teleinfo2mqtt:
    image: fmartinou/teleinfo2mqtt
    container_name: teleinfo2mqtt
    devices:
      - /dev/ttyUSB1:/dev/ttyUSB1
    environment:
      - SERIAL=/dev/ttyUSB1
      - MQTT_BASE_TOPIC=teleinfo
      - MQTT_URL=${MQTT_URL}
      - MQTT_USER=${MQTT_USER}
      - MQTT_PASSWORD=${MQTT_PASSWORD}
      - EMIT_INTERVAL=30
      - TZ=${TZ} 
      - TIC_MODE=standard
      - HASS_DISCOVERY=true
      - HASS_DISCOVERY_PREFIX=homeassistant
    restart: unless-stopped
```
en repassant en 8.0.2 pas de problème il prends bien ttyUSB1

fmartinou, 2022-09-25:
Arf, je vois le bug rien qu'en relisant le code 😔.

Je vais fixer.

En attendant le correctif, il est possible de monter le device ttyUSB1 en tant que ttyUSB0 dans le conteneur. Ça devrait fonctionner.

```yaml
devices:
     - /dev/ttyUSB1:/dev/ttyUSB0
```

fmartinou, 2022-09-26:
C'est normalement corrigé 🤞 
Et releasé (`8.1.1`).

closed 2022-09-26 by fmartinou