fmartinou/teleinfo2mqtt #35: Republier configuration Home Assistant on reconnect

rtorrente, 2022-09-28:
En cas de coupure du serveur MQTT, si pour une raison quelconque au moment de la re-connexion, le serveur MQTT n'a pas de persistance des messages retain (Ce qui est le cas par défaut si le fichier de persistance n'est pas configuré dans le serveur mosquitto), il n'y aura plus de configuration auto-discovery home-assistant sur le broker.

En cas de reboot de home-assistant, la teleinfo paraîtra indisponible jusqu'à un reboot du container teleinfo2mqtt qui publiera à nouveau la configuration home-assistant. 

Je pense qu'il suffit de passer discoveryConfigurationPublished à false au moment de l'évènement  "reconnect" pour avoir le résultat attendu

fmartinou, 2022-10-15:
Hi @rtorrente 

Your issue has just been fixed in the latest `8.1.2` version.

fmartinou, 2022-10-20:
J'ai dû rollback à cause de régressions chez certains utilisateurs que je ne m'explique pas pour le moment.

fmartinou, 2022-10-24:
next try with `fmartinou/teleinfo2mqtt:fix__35` 🤞