fmartinou/teleinfo2mqtt #36: erreur lancement sur rpi4

gabberhq, 2022-10-16:
bonjour 

lorsque je suis les instructions pour lancer l'image en ayant pris soin de renseigner les paramètres ( URL MQTT et port USB ) il ne se passe rien et dans les logs j'obtiens l'erreur suivante : 

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbee25234

#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbedea204
Illegal instruction (core dumped)
Illegal instruction (core dumped)

fmartinou, 2022-10-16:
Bonjour, c'est une erreur que vous n'aviez pas sur une ancienne version ou bien c'est la première fois que vous essayez de faire tourner cette app ?

fmartinou, 2022-10-16:
Essayez en spécifiant cette option supplémentaire lors du lancement du conteneur docker

`--security-opt seccomp=unconfined`

fmartinou, 2022-10-16:
Ou ça ?
https://www.zigbee2mqtt.io/guide/installation/02_docker.html#notes-for-raspberry-pi-users

gabberhq, 2022-10-16:
nikel , ca fonctionne, merci beaucoup

en revanche, passer du plugin téléinfo de jeedom à un decodage des données brutes ne va pas etre une mince affaire il me semble

fmartinou, 2022-10-16:
> nikel , ca fonctionne, merci beaucoup

De spécifier l'options `--security-opt` ou bien d'installer la lib `libseccomp2` ?

(afin que je rajoute ça dans la doc)

> en revanche, passer du plugin téléinfo de jeedom à un decodage des données brutes ne va pas etre une mince affaire il me semble

Les quelques données intéressantes sont toujours les mêmes (les indexes `BASE`, `HCHC`, `HCHP`, la conso instantanée `PAPP`, etc.).

La plupart des utilisateurs de `teleinfo2mqtt` utilise Home-Assistant et l'intégration est plutôt facile et fluide.
Vous utilisez jeedom ? Si oui, par curiosité, pourquoi ne plus utiliser le plugin officiel ?

gabberhq, 2022-10-16:
le plugin officiel est fort complet et offre toute les informations nécessaires, c'est vrai. En revanche, j'aurais préféré migrer vers une centralisation MQTT de l'ensemble des données et à terme probablement vers Home-Assistant. D'où la volonté d'implementer votre solution qui me permettrait de conserver le modem teleinfo usb plutôt que de basculer sur une solution à base de esp8266 /tasmota / esphome :)

gabberhq, 2022-10-16:
c'est l'installation de la lib qui a permis de corriger le problème. Par ailleurs, preciser "localhost" pour le serveur MQTT ne fonctionne pas, il faut preciser l'ip exacte de la machine.

fmartinou, 2022-10-16:
Merci pour la précision ; je vais compléter la doc 👍

> Par ailleurs, preciser "localhost" pour le serveur MQTT ne fonctionne pas, il faut preciser l'ip exacte de la machine.

Tout dépend de comment on lance le conteneur. Sur le réseau "bridge" par défaut, effectivement, localhost ça ne va pas marcher.
Sur le network "host", ça fonctionnera si le broker mqtt est sur le même hôte.

Je peux améliorer la doc sur ce point aussi.

gabberhq, 2022-10-16:
je viens de regarder les trames enregistées et n'y retrouve aucune informations claires, notamment celles relatives à la conso instantannée, totale de la veille , etc . Peut-etre est-ce lié au fait que je suis en triphasé ?

fmartinou, 2022-10-16:
Normalement, les données sont juste celles reçues par la teleinfo et renvoyées telles quelles.

Vous êtes en mode standard ou historique ?
Vous avez un example de message mqtt ?

gabberhq, 2022-10-16:
je suis en mode standard.

voici la trame MQTT ( j'ai supprimé le numéro du compteur ) : 

{"ADSC":{"raw »: »XXXXXX », »value »:XXXXX}, »VTIC":{"raw":"02","value":2},"DATE":{"raw":"","value":"","timestamp":{"dst":"summer","date":"2022-10-16T18:15:02.000Z"}},"NGTF":{"raw":"H PLEINE/CREUSE ","value":"H PLEINE/CREUSE "},"LTARF":{"raw":" HEURE  PLEINE  ","value":" HEURE  PLEINE  "},"EAST":{"raw":"020966534","value":20966534},"EASF01":{"raw":"010485700","value":10485700},"EASF02":{"raw":"010373522","value":10373522},"EASF03":{"raw":"000107312","value":107312},"EASF04":{"raw":"000000000","value":0},"EASF05":{"raw":"000000000","value":0},"EASF06":{"raw":"000000000","value":0},"EASF07":{"raw":"000000000","value":0},"EASF08":{"raw":"000000000","value":0},"EASF09":{"raw":"000000000","value":0},"EASF10":{"raw":"000000000","value":0},"EASD01":{"raw":"004248863","value":4248863},"EASD02":{"raw":"003931974","value":3931974},"EASD03":{"raw":"006261367","value":6261367},"EASD04":{"raw":"006524330","value":6524330},"IRMS1":{"raw":"000","value":0},"IRMS2":{"raw":"001","value":1},"IRMS3":{"raw":"001","value":1},"URMS1":{"raw":"238","value":238},"URMS2":{"raw":"236","value":236},"URMS3":{"raw":"234","value":234},"PREF":{"raw":"15","value":15},"PCOUP":{"raw":"15","value":15},"SINSTS":{"raw":"00417","value":417},"SINSTS1":{"raw":"37059","value":37059},"SINSTS2":{"raw":"73569","value":73569},"SINSTS3":{"raw":"07131","value":7131},"SMAXSN":{"raw":"03190","value":3190,"timestamp":{"dst":"summer","date":"2022-10-16T04:09:32.000Z"}},"SMAXSN1":{"raw":"00060","value":60,"timestamp":{"dst":"summer","date":"2022-10-16T08:56:55.000Z"}},"SMAXSN2":{"raw":"02340","value":2340,"timestamp":{"dst":"summer","date":"2022-10-16T12:42:16.000Z"}},"SMAXSN3":{"raw":"03040","value":3040,"timestamp":{"dst":"summer","date":"2022-10-16T04:10:09.000Z"}},"SMAXSN-1":{"raw":"04430","value":4430,"timestamp":{"dst":"summer","date":"2022-10-15T15:24:53.000Z"}},"SMAXSN1-1":{"raw":"00970","value":970,"timestamp":{"dst":"summer","date":"2022-10-15T15:05:19.000Z"}},"SMAXSN2-1":{"raw":"02500","value":2500,"timestamp":{"dst":"summer","date":"2022-10-15T21:05:28.000Z"}},"SMAXSN3-1":{"raw":"04270","value":4270,"timestamp":{"dst":"summer","date":"2022-10-15T15:24:53.000Z"}},"CCASN":{"raw":"00223","value":223,"timestamp":{"dst":"summer","date":"2022-10-16T18:00:00.000Z"}},"CCASN-1":{"raw":"00193","value":193,"timestamp":{"dst":"summer","date":"2022-10-16T17:00:00.000Z"}},"UMOY1":{"raw":"239","value":239,"timestamp":{"dst":"summer","date":"2022-10-16T18:10:00.000Z"}},"UMOY2":{"raw":"233","value":233,"timestamp":{"dst":"summer","date":"2022-10-16T18:10:00.000Z"}},"UMOY3":{"raw":"233","value":233,"timestamp":{"dst":"summer","date":"2022-10-16T18:10:00.000Z"}},"STGE":{"raw":"003A4401","value":3},"MSG1":{"raw":"     PAS DE          MESSAGE    ","value":"     PAS DE          MESSAGE    "},"PRM":{"raw":"05669464533754","value":5669464533754},"RELAIS":{"raw":"000","value":0},"NTARF":{"raw":"02","value":2},"NJOURF":{"raw":"00","value":0},"NJOURF+1":{"raw":"00","value":0},"PJOURF+1":{"raw":"00008002 0204C001 07048002 0E04C001 11048002 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE","value":8002}}

la conso totale du jour doit etre calculée ainsi que celle de la veille je crois.

fmartinou, 2022-10-16:
Toutes les étiquettes de la teleinfo sont documentées par Enedis dans ce document

https://www.enedis.fr/media/2027/download&ved=2ahUKEwjtxfLuruX6AhUG_hoKHYSuC3MQFnoECBkQAQ&usg=AOvVaw32Y35kiJxzRPJk9R-jrOzv

gabberhq, 2022-10-16:
merci, je vais regarder tout ca de près

fmartinou, 2023-03-12:
Je cloture l'issue pour cause d'inactivité ; n'hésitez pas à la rouvrir s'il y a besoin.

closed 2023-03-12 by fmartinou