fmartinou/teleinfo2mqtt #37: feat request : using uptime kuma it would be great to publish some success message on topic

ka2er, 2022-10-16:
Hi,

I'm using your software. Great works thanks (merci) !

I'm also using kuma uptime (https://github.com/louislam/uptime-kuma) to track that all my homelab services are alive.
This status solution is able to track a success message under a MQTT topic to gather wether or not the client publishing here is alive.
Having just a boolean refreshed like 

```teleinfo/up = true```

would fit the case. If the message get too old and don't get refreshed it would indicates that the client isn't publishing anymore and that we have to bother about our teleinfo stack.

At the time being kuma isn't able to do more elaborate check thant strict equality (AFAIK) so I'm trying to ask here if you mind accept this feature request.

If so, I believe I could send you a merge request as it seems like a very simple enhancement to implement + test.

Just tell me if I you are willing to accept this kind of PR.

fmartinou, 2022-10-16:
Hi,

Thank you for your support 👍!

wud already exposes its [health via http](https://fmartinou.github.io/whats-up-docker/#/monitoring/).

So I would say that Kuma Uptime can already monitor it through http... doesn't it?

ka2er, 2022-10-16:
I don't get the dependency on WUD as I'm not using at the time being.
I just check it seems to be a project to ensure our running docker image are up to date.

In the case of teleinfo2mqtt I don't see how WUD can state that teleinfo data are still getting pushed to MQTT and application is alive (in terms of data and not container still running).

But I have just check quickly maybe I miss something ?

fmartinou, 2022-10-16:
Sorry 😅

I confused 2 projects I'm maintaining 😄.

Yes, providing a health status through mqtt is a great idea 👍.

It won't be very hard to implement.
But if you can help with a PR; even better,  you're welcome 🙂!

closed 2022-11-27 by fmartinou

fmartinou, 2022-11-27:
Release as `8.2.0`

ka2er, 2022-11-27:
tested and it does the job thanks !!

[image]

fmartinou, 2022-11-27:
Great!
Thank you very much for your contribution!

commit 2b63ba05 (2022-11-27): fix #37 : add mqtt health topic (#38)

Add mqtt health topic
commit a623bd7d (2022-11-27): fix #37 : add mqtt health topic (#38)

Add mqtt health topic
commit 8dc558ba (2022-11-27): fix #37 : add mqtt health topic (#38)

Add mqtt health topic