fmartinou/teleinfo2mqtt #42: Pas de remontée des téléinformations de production

Fox33290, 2023-01-19:
Bonjour.
J'utilise un module un module Téléinfo USB Rail DIN
J'utilise teleinfo2mqtt version: 8.2.0 en addon sous home assistant 2023.1.5.
Compteur Linky en mode standard, les informations de consommations remontent correctement.
Par contre, je n'ai pas accès aux informations de production (SINTI ....). Elles n'apparaissent pas dans les entités Téléinfo. 
Je précise que je suis en auto production sans revente. 
Quelqu'un a t'il testé dans cette configuration ?

(Première fois sur ce type de "forum" et un peu novice en domotique, excusez moi si je suis maladroit)
Merci d'avance.

spamoi33, 2023-01-19:
Bonjour,

C'est "normal". L'index d'injection est affiché sur l'écran LCD, mais ne remonte pas par la TIC (SINSTI) si le compteur n'est pas en mode "production".

https://forum-photovoltaique.fr/viewtopic.php?t=43906

J'ai eu cette question car j'ai 2 panneaux Sunology ;)

Je ne sais pas à qui il faut faire la demande, j'avais envoyé un mail à mon fournisseur mais pas de réponse...
Pour le moment je monitor ma production solaire avec une prise connectée et cette intégration : https://www.home-assistant.io/integrations/integration/

Fox33290, 2023-01-19:
OK merci. J'avais lu ça aussi quelque part.
J'ai validé ma production chez Enedis aujourd'hui. Je crois que les Linky sont mis à jour la nuit. Je verrai demain s'il ont passé mon compteur en mode Production. Si ce n'est pas le cas, je verrai avec eux comment faire.

spamoi33, 2023-01-19:
Tu peux me passer le mail stp ? je tenterai également :)
Et oui, la maj se fait à 00h ^^

Fox33290, 2023-01-19:
Je n'ai pas de mail. En fait, j'ai fait une demande de raccordement sur le site Enedis Portail Raccordement, fourni les documents nécessaires et il ne me restait plus qu'à valider le démarrage de ma production. On peut penser qu'il me considère à partir d'aujourd'hui comme "Producteur", même si je ne revends rien. Je verrai demain si le Linky a évolué.

fmartinou, 2023-01-28:
> Je n'ai pas de mail. En fait, j'ai fait une demande de raccordement sur le site Enedis Portail Raccordement, fourni les documents nécessaires et il ne me restait plus qu'à valider le démarrage de ma [...]

Je suis exactement dans la même situation : autoconsommation sans revente depuis septembre dernier.

Cela n'a provoqué aucun changement sur mon Linky qui est resté en mode historique.

J'ai démarché mon fournisseur (Vattenfall) pour le faire passer en mode standard... toujours pas de réponse à ce jour.

Pour le moment, je suis tout ça avec des pinces ampèremétriques(Shelly EM/3EM) et ça marche tout aussi bien ^^.

Fox33290, 2023-01-28:
J'ai eu Enedis au téléphone. En autoproduction sans revente, ils ne nous considèrent pas comme producteur, du coup, le Linky ne remonte pas dans la téléinfo les paramètres de production (SINTI...). Pourtant, il l'affiche sur son afficheur, mon compteur étant en mode Standard. C'est assez bizarre comme situation. La personne que j'ai eu au téléphone m'a dit qu'elle faisait remonter ce cas particulier au niveau national. Elle doit me recontacter. A suivre....

spamoi33, 2023-01-28:
C’est bien décrit comme ça dans la doc, il n’y a rien de bizarre ! Il faut être en mode producteur pour que ça remonte dans la TIC c’est tout. 

Certains ont réussi à le faire même en auto-consommation, ça doit dépendre de sûr qui on tombe :/

fmartinou, 2023-03-12:
Je cloture l'issue pour cause d'inactivité ; n'hésitez pas à la rouvrir s'il y a besoin.

closed 2023-03-12 by fmartinou