fmartinou/teleinfo2mqtt #50: Mode standard / home assistant mauvaise valeur pour STGE

alexstaz, 2023-03-15:
Bonjour,

J'utilise le add on de home assistant qui lit mon Linky en mode standard. Il lit correctement les valeurs, mais pour STGE, la valeur est transformée alors que cela devrait etre de l'hexadecimal, donc du texte.
Ex (valeur prise dans les topics de MQTT)
STGE:
  raw: 003A0001
  value: 3

J'ai du mal a comprendre comment on se retrouve juste avec 3 ? Ca tronque juste avant le premier caractère ?
Une idée de comment corriger ca ?

Merci

fmartinou, 2023-03-26:
Bonjour,

L'appli fait de la coercion de types en "best effort" pour retourner des valeurs mieux typées que le `string` récupérées.
[Voir le code ici](https://github.com/fmartinou/teleinfo2mqtt/blob/master/app/teleinfo/TicMode.js#L194-L195)

Dans le cas présent
```js
const valueNumber = Number.parseInt('003A0001', 10); // -> return 3
```

La valeur originale non manipulée reste accessible via la propriété `raw` ; vous ne pouvez pas l'utiliser ?

closed 2023-07-05 by fmartinou

fmartinou, 2023-07-05:
Released as `8.3.1`.