fmartinou/teleinfo2mqtt #57: Wrong device class for PAPP sensor

NathanKun, 2023-08-05:
I got a warning in log:
Entity sensor.linky_032064323057_teleinfo_xxx_papp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'VA' which is not a valid unit for the device class ('power') it is using; expected one of ['W', 'kW'];

This is related to the PAPP sensor, which is configured as a 'power' device class, but using 'VA' as unit.
I think we should use  the 'apparent_power' device class for the PAPP sensor:
https://www.home-assistant.io/integrations/sensor/

fmartinou, 2023-08-26:
Hi,

Thank your for reporting this issue; I didn't know about the `apparent_power`.
Your analysis seems right; I'll make the change.

fmartinou, 2023-08-26:
Released as `8.3.3`

closed 2023-08-26 by fmartinou