fmartinou/teleinfo2mqtt #60: Coupure sur pjourf+1 en mode standard

WorldOfGZ, 2023-09-01:
Bonjour

Je viens d'installer cet addon sur un HA à jour. 
Version de l'addon 8.3.3

J'ai visiblement un traitement sur le label PJOUR+1. A chaque fois il fait déconnecte le MQTT.
Sur le log MQTT j'ai cette info : Client mqttjs_153f3f22 disconnected due to malformed packet.

Sur le log en mode info, j'ai ces renseignements : 
`20:22:46.849  INFO teleinfo2mqtt:
    Starting teleinfo2mqtt with configuration = {
      emitInterval: 10,
      hassDiscovery: true,
      hassDiscoveryPrefix: 'homeassistant',
      logLevel: 'info',
      mqttBaseTopic: 'teleinfo',
      mqttUrl: 'mqtt://ip:1883',
      mqttUser: 'user',
      mqttPassword: '<hidden>',
      mqttTlsClientKey: undefined,
      mqttTlsClientCert: undefined,
      mqttTlsCaChain: undefined,
      mqttTlsRejectUnauthorized: true,
      serial: '/dev/ttyUSB0',
      ticMode: 'standard'
    }
20:22:46.850  INFO teleinfo2mqtt: Connecting to port [/dev/ttyUSB0] with standard TIC mode
20:22:46.852  INFO teleinfo2mqtt: Connected to port [/dev/ttyUSB0]
20:22:46.853  INFO teleinfo2mqtt: Connecting to MQTT broker [mqtt://ip:1883]
20:22:46.885  INFO teleinfo2mqtt: Connected to MQTT broker [mqtt://ip:1883]
20:22:46.887  WARN teleinfo2mqtt: Corrupted line received [243	R
]
20:22:58.172  INFO teleinfo2mqtt: Publish configuration for tag ADSC for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_adsc/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag VTIC for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_vtic/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag DATE for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_date/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag NGTF for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_ngtf/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag LTARF for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_ltarf/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EAST for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_east/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF01 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf01/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF02 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf02/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF03 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf03/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF04 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf04/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF05 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf05/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF06 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf06/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF07 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf07/config]
20:22:58.173  INFO teleinfo2mqtt: Publish configuration for tag EASF08 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf08/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag EASF09 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf09/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag EASF10 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easf10/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag EASD01 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easd01/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag EASD02 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easd02/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag EASD03 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easd03/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag EASD04 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_easd04/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag IRMS1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_irms1/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag URMS1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_urms1/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag PREF for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_pref/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag PCOUP for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_pcoup/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag SINSTS for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_sinsts/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag SMAXSN for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_smaxsn/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag SMAXSN-1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_smaxsn-1/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag CCASN for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_ccasn/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag CCASN-1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_ccasn-1/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag UMOY1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_umoy1/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag STGE for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_stge/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag MSG1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_msg1/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag PRM for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_prm/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag RELAIS for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_relais/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag NTARF for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_ntarf/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag NJOURF for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_njourf/config]
20:22:58.174  INFO teleinfo2mqtt: Publish configuration for tag NJOURF+1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_njourf+1/config]
20:22:58.175  INFO teleinfo2mqtt: Publish configuration for tag PJOURF+1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_pjourf+1/config]
20:22:59.183  INFO teleinfo2mqtt: Reconnecting to the MQTT broker...`

J'ai eu un peu de mal, ça donne cette ligne en mode debug : 
`20:19:48.426 DEBUG teleinfo2mqtt: Raw frame [PJOURF+1	00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE	9
]
20:19:48.426 DEBUG teleinfo2mqtt: Split frame [PJOURF+1,00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE,9
]
20:19:48.426 DEBUG teleinfo2mqtt: Value for label PJOURF+1 = 00008001 NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE NONUTILE
20:19:48.426 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]`

Je sais pas trop si ça vient de chez moi ou non, si je peux avoir un petit coup de main svp !

fmartinou, 2023-09-02:
Bonjour,

Je ne suis pas absolument certain que le problème soit lié à l'étiquette PJOURF+1.

```
20:22:58.175 INFO teleinfo2mqtt: Publish configuration for tag PJOURF+1 for discovery to topic [homeassistant/sensor/teleinfo/XXXXXXXXXXXX_pjourf+1/config]
20:22:59.183 INFO teleinfo2mqtt: Reconnecting to the MQTT broker...
```
Il y a 8ms entre les 2 lignes, donc ces 2 logs ne sont pas obligatoirement à correller.

Si tu regardes le contenu des topics mqtt, tu vois des messages pour toutes les autres étiquettes sauf PJOURF+1 ?

WorldOfGZ, 2023-09-02:
Bonjour

Le dernier message que je reçoit sur mqtt concerne njourf
Je n'ai pas de trace de pjourf+1, mais pas non plus de njourf+1

J'utilise mosquitto dans l'addon de HA, la version semble être selon le changelog la 2.0.17

Mounicq, 2023-09-29:
Je viens de constater aussi le problème, le caractère '+' n'est pas autorisé dans les topics mqtt.
mqtt/hass.js 
L42:
-        const discoveryTopic = `${hassDiscoveryPrefix}/sensor/${mqttBaseTopic}/${id}_${label.toLowerCase()}/config`;
+        const discoveryTopic = `${hassDiscoveryPrefix}/sensor/${mqttBaseTopic}/${id}_${label.replace('+', '_plus_').toLowerCase()}/config`;

Makokoy79, 2023-11-03:
Bonsoir à tous,
J'ai le même problème? Ce que je vois dans la doc de Enedis ici https://www.enedis.fr/media/2035/download c'est qu'en fonction du compteur ce ne sont pas les mêmes informations qui sont remontées, ou pas forcément sous la même forme.
Je début sous HA, mais j'aimerai aider. Dans quel fichier les trames reçues sont décodées et formatées ?

Makokoy79, 2023-11-10:
Si cela peut aider, voici le journal des messages reçus et parsés par teleinfo2mqtt

`15:27:37.553 DEBUG teleinfo2mqtt: Split frame [EASD02,019555407,E
]
15:27:37.553 DEBUG teleinfo2mqtt: Value for label EASD02 = 019555407
15:27:37.553 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.577 DEBUG teleinfo2mqtt: Raw frame [EASD03	003031843	8
]
15:27:37.577 DEBUG teleinfo2mqtt: Split frame [EASD03,003031843,8
]
15:27:37.577 DEBUG teleinfo2mqtt: Value for label EASD03 = 003031843
15:27:37.577 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.593 DEBUG teleinfo2mqtt: Raw frame [EASD04	004670293	B
]
15:27:37.594 DEBUG teleinfo2mqtt: Split frame [EASD04,004670293,B
]
15:27:37.594 DEBUG teleinfo2mqtt: Value for label EASD04 = 004670293
15:27:37.594 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.610 DEBUG teleinfo2mqtt: Raw frame [IRMS1	006	4
]
15:27:37.610 DEBUG teleinfo2mqtt: Split frame [IRMS1,006,4
]
15:27:37.610 DEBUG teleinfo2mqtt: Value for label IRMS1 = 006
15:27:37.610 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.624 DEBUG teleinfo2mqtt: Raw frame [URMS1	239	H
]
15:27:37.624 DEBUG teleinfo2mqtt: Split frame [URMS1,239,H
]
15:27:37.624 DEBUG teleinfo2mqtt: Value for label URMS1 = 239
15:27:37.624 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.639 DEBUG teleinfo2mqtt: Raw frame [PREF	12	B
]
15:27:37.639 DEBUG teleinfo2mqtt: Split frame [PREF,12,B
]
15:27:37.639 DEBUG teleinfo2mqtt: Value for label PREF = 12
15:27:37.639 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.659 DEBUG teleinfo2mqtt: Raw frame [PCOUP	12	\
]
15:27:37.659 DEBUG teleinfo2mqtt: Split frame [PCOUP,12,\
]
15:27:37.659 DEBUG teleinfo2mqtt: Value for label PCOUP = 12
15:27:37.659 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.695 DEBUG teleinfo2mqtt: Raw frame [SINSTS	01331	N
]
15:27:37.695 DEBUG teleinfo2mqtt: Split frame [SINSTS,01331,N
]
15:27:37.695 DEBUG teleinfo2mqtt: Value for label SINSTS = 01331
15:27:37.695 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.733 DEBUG teleinfo2mqtt: Raw frame [SMAXSN	H231110065319	04400	5
]
15:27:37.733 DEBUG teleinfo2mqtt: Split frame [SMAXSN,H231110065319,04400,5
]
15:27:37.733 DEBUG teleinfo2mqtt: Value for label SMAXSN = 04400
15:27:37.733 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.768 DEBUG teleinfo2mqtt: Raw frame [SMAXSN-1	H231109090852	05140	]
]
15:27:37.768 DEBUG teleinfo2mqtt: Split frame [SMAXSN-1,H231109090852,05140,]
]
15:27:37.768 DEBUG teleinfo2mqtt: Value for label SMAXSN-1 = 05140
15:27:37.768 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.805 DEBUG teleinfo2mqtt: Raw frame [CCASN	H231110150000	01075	6
]
15:27:37.805 DEBUG teleinfo2mqtt: Split frame [CCASN,H231110150000,01075,6
]
15:27:37.805 DEBUG teleinfo2mqtt: Value for label CCASN = 01075
15:27:37.805 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:37.837 DEBUG teleinfo2mqtt: Raw frame [CCASN-1	H231110140000	01103	K
]
15:27:37.837 DEBUG teleinfo2mqtt: Split frame [CCASN-1,H231110140000,01103,K
]
[... 211 lines cut ...]
15:27:44.788 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:44.810 DEBUG teleinfo2mqtt: Raw frame [LTARF	  HEURE  PLEINE 	A
]
15:27:44.810 DEBUG teleinfo2mqtt: Split frame [LTARF,  HEURE  PLEINE ,A
]
15:27:44.810 DEBUG teleinfo2mqtt: Value for label LTARF =   HEURE  PLEINE 
15:27:44.810 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:44.834 DEBUG teleinfo2mqtt: Raw frame [EAST	034557076	4
]
15:27:44.834 DEBUG teleinfo2mqtt: Split frame [EAST,034557076,4
]
15:27:44.834 DEBUG teleinfo2mqtt: Value for label EAST = 034557076
15:27:44.834 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:44.858 DEBUG teleinfo2mqtt: Raw frame [EASF01	010331379	=
]
15:27:44.858 DEBUG teleinfo2mqtt: Split frame [EASF01,010331379,=
]
15:27:44.858 DEBUG teleinfo2mqtt: Value for label EASF01 = 010331379
15:27:44.858 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]
15:27:44.882 DEBUG teleinfo2mqtt: Raw frame [EASF02	024225697	H
]
15:27:44.882 DEBUG teleinfo2mqtt: Split frame [EASF02,024225697,H
]
15:27:44.882 DEBUG teleinfo2mqtt: Value for label EASF02 = 024225697
15:27:44.882 DEBUG teleinfo2mqtt: Frame parsed [[object Object]]`

Makokoy79, 2023-11-11:
Re :-)
Autre information que je n'avais pas vue dans le journal de mosquitto :

2023-11-11 17:34:06: New client connected from 172.30.32.1:52292 as mqttjs_1997842f (p2, c1, k60, u'mqtt').
2023-11-11 17:34:16: Client mqttjs_1997842f disconnected due to malformed packet.

IL s'agirait bien apparemment d'une trame mal interprétée.

closed 2023-12-09 by fmartinou

fmartinou, 2023-12-09:
Released as `9.0.3`.