fmartinou/teleinfo2mqtt #64: Demande d'aide suite passage TIC en mode standard

florianroulet, 2023-11-16:
Bonjour, 

Je viens de faire passer récemment ma TIC en mode standard et cela a complètement "cassé" ma relève de téléinfo.
J'ai un pi0 sur lequel tourne docker connecté via un module [téléinformation GCE Electronics](https://www.gce-electronics.com/fr/carte-electronique-oem-relais-usb/655-module-teleinfo-usb.html)

Je ne pense pas que le problème vienne de teleinfo2mqtt car même en faisant un "cat /dev/ttyUSB0" j'ai un affichage tout cassé.

J'avoue que je suis un peu perdu. Avant le passage en mode standard, tout marchait bien.

```
cat /dev/ttyUSB0
 8791040B
AASD0000000   @�9
AS0` 0`  0` 0 HPHV2	004	2
TRM1	02REF	0H     PBN@@ B
                             S	0082X
SL@XV\P230004  09!0770	
XS

CC`0& 00  b`n18	000008
  	H23110406" `0   &0	@
                                CCASN-1	H2200017000001404
TLNX0  0#   d17400 	 "0    FE	 13A401	B
MSG0	AR @@      
                   @SF@    @   <
PAH@@B `09712600B92080.
N@RF00	  O
JOUPF"   	B
                PJOURF1	00004001  40   02 22004000 NLQP@H@ NONATILE OUTHH@ ONE NONWHL@

                                                                                      T 
                                                                                        NN@HELNLILE	.

AS	061860667428@

DATE	H231116060049B
LTRF  HPBHBT @A 810434@BD01 "0 04726	D

DASF00  2530&&&
ESD04 0 00  0`00  
             @@BD``	000000000,	@@BD`l`0000000'
F08L7	0000000@F 8	   0 0DA 0	00000000  
ASD030 `000000000	"

D@R@00	0 0  00 0
PEF1	0HBNPP0rD@Hf ""BB
IAPRL@d31!16010916      06644	C083V
RH@XS
     ,1@23  15 "4&5808490(
CB@RL@`fb116173000
0116@
O03100406000` "     	X
F01"04  	C	H`
S @@      MESGD      8
PR
  	09712200:2000 dIS	000
                                   B
NT@RL	00
OD	  	
JOURF*00000400006      00004000 LN\�P@H@
                                        NLE NONUTILD LNUTAL@LNUPILE NNHLL
                                                                          
                                                                          LTALE	,
TICC	02J0<035'$"8	A
D@T@@ 31! $174248

HPF  @PT      L +
ESF002810462D	&

ASF000000000AA 40   0   0
AVF5	    00000	&

D@RL06	      00'
ESB0000000000@@&
                08	000000000	(

E  0 0  08 7910640000000C 
D@R@0`	00`0  339
                 :
EASD06
EASD04``0000 0   	
HS 002

@DIC! ""	B

PRE08
SISTS`1 00824W
RH@PB\	H""  16010:206600	       RH@p
                                           )b0f 11  3 & 09080,

FCA@0" 116!&"000	00178	D
                                 CCARL,0@`fb00$ 70000
                                                     0140X
IOP0@ 311!& &   `"34	1	SD	  "A  01B
HRF0	PS DE          H@BFB       	<

PPH0970 "  0p2082	7
REL@H000	

N@RF 0	0"

@OUPF
PJOURF+100B     00004000 040 h0  ``004001 NONTTILD LNLXXALD LN\HTILNONUP@LE NONTTJLE LN 
                                                                                        NN@HENONTILE	.
                                                                                                         ADRF
04104020>430D@IC "J
NFTE	    TDLPN     D
L`  EU   A	08104640$

D@VL0! "0  h726	E
EF004538666H
ESD02 `````000	$
EASF0  0`  0` A``	000000000,^C

```

```
docker run --name teleinfo2mqtt --restart always  --device=/dev/ttyUSB0:/dev/ttyUSB0   -e MQTT_URL=mqtt://X.X.X.X:1883 -e MQTT_USER=homeassistant -e MQTT_PASSWORD=leijoo3Vu3ohhahdacahYohriegh7pha2ogi6ainieSohx9xeedai8Echoh0oodu -e TIC_MODE=standard fmartinou/teleinfo2mqtt
16:34:20.669  INFO teleinfo2mqtt:
    Starting teleinfo2mqtt with configuration = {
      emitInterval: 10,
      hassDiscovery: true,
      hassDiscoveryPrefix: 'homeassistant',
      logLevel: 'info',
      mqttBaseTopic: 'teleinfo',
      mqttUrl: 'mqtt://X.X.X.X:1883',
      mqttUser: 'homeassistant',
      mqttPassword: '<hidden>',
      mqttTlsClientKey: undefined,
      mqttTlsClientCert: undefined,
      mqttTlsCaChain: undefined,
      mqttTlsRejectUnauthorized: true,
      serial: '/dev/ttyUSB0',
      ticMode: 'standard'
    }
16:34:20.737  INFO teleinfo2mqtt: Connecting to port [/dev/ttyUSB0] with standard TIC mode
16:34:20.839  INFO teleinfo2mqtt: Connected to port [/dev/ttyUSB0]
16:34:20.906  INFO teleinfo2mqtt: Connecting to MQTT broker [mqtt://192.168.0.32:1883]
16:34:22.621  WARN teleinfo2mqtt: Invalid value received for labelSMASN-1 [H260110020. p]
]6:34:22.638  WARN teleinfo2mqtt: Corrupted line received [PRM	0860020011" 0 &
16:34:22.646  WARN teleinfo2mqtt: Corrupted line received [DL@S	  0
                                                                   ]
16:34:22.648  WARN teleinfo2mqtt: Invalid value received for label NT@RB [02]
]6:34:22.650  WARN teleinfo2mqtt: Corrupted line received [JND1	00
16:34:22.656  WARN teleinfo2mqtt: Invalid value received for label PJGURF+1 [00004001 0600402 0000001 LNLPPILE LNL@@ILEN
                                                                                                                        D
                                                                                                                         NIL ONUTHH@NONUTILD OUTHH@NONE]
]6:34:22.658  WARN teleinfo2mqtt: Corrupted line received [ADR0<`0& " & 8	A
16:34:22.659  WARN teleinfo2mqtt: Corrupted line received [VTIB	0H
                                                                  ]
]6:34:22.661  WARN teleinfo2mqtt: Corrupted line received [D@XE	H0200 617340$E
]6:34:22.662  WARN teleinfo2mqtt: Corrupted line received [N M     F
16:34:22.663  WARN teleinfo2mqtt: Corrupted line received [LT@PF	    @@ BHDU    
]
ESF0103850462FWARN teleinfo2mqtt: Corrupted line received [EAR08104"   	&
              ]
16:34:22.686  WARN teleinfo2mqtt: Invalid value received for label E@RD000800200pp� [@0f 0`  0  0
                                                                                                DBL04]
16:34:22.687  WARN teleinfo2mqtt: Corrupted line received []
16:34:22.689  WARN teleinfo2mqtt: Invalid value received for label @RD 8 [  0` 00 A@B`2]
]6:34:22.690  WARN teleinfo2mqtt: Corrupted line received [
]6:34:22.691  WARN teleinfo2mqtt: Corrupted line received [D@RD 0	0      0"
16:34:22.692  WARN teleinfo2mqtt: Invalid value received for label EASD01 [0806800@@R00 0  ``50&`200`  0` 0]
16:34:22.694  WARN teleinfo2mqtt: Invalid value received for label IRI  [002]
16:34:22.695  WARN teleinfo2mqtt: Corrupted line received [URMR02PTEF 0@]
C6:34:22.715  WARN teleinfo2mqtt: Invalid value received for label PCOD@ 0
XSN/2]11 &  0r `&64`
16:34:22.716  WARN teleinfo2mqtt: Corrupted line received [SDE	01A000B

```

fmartinou, 2023-11-19:
Bonjour,

Quand tu `cat` directement le device usb et que tu observes que c'est aussi cassé, tu as bien changé la conf associée pour le mode standard ?
(baud rate qui passe de 1200 à 9600 par exemple...)

florianroulet, 2023-11-21:
Bonjour, 
Je reviens avec des nouvelles ! 
Tout d'abord, pour répondre à la question, oui, j'avais bien passé mon COM en 9600 bauds. C'est pourquoi je trouvais très étrange que cela ne fonctionne pas. 
En cherchant un peu plus (je n'ai plus le lien malheureusement) je suis tombé sur message d'une personne disant que les anciens modules GCE ne fonctionnaient qu'en 1500 bauds. J'en ai donc commandé un neuf, le mien datant de 2015.
Et magie ! 
Tout fonctionne ! le cat, le containeur, les messages reçus sur HA.
Bref, un grand merci pour le temps pour la réponse et le dev sur le soft !

fmartinou, 2023-11-21:
Ravi de voir que tout fonctionne parfaitement 👌.

closed 2023-11-21 by fmartinou