fmartinou/teleinfo2mqtt #70: Plus de remonté de données et break des entity_id

pierrepinon, 2023-11-27:
Bonsoir,

Depuis la dernière mise à jour, l'ensemble des sensors ont changés de nom et malgré les publications sur le broker MQTT, aucun sensor ne sont mise à jour...

pierrepinon, 2023-11-27:
J'ai beaucoup d'erreur de cast dans les logs

`ValueError: could not convert string to float: "''"`

Possible de rollback ? Car là c'est inutilisable en l'état pour tout le monde à mon avis

fmartinou, 2023-11-27:
Je vais rollback la version référencée sur hass.io le temps d'y voir plus clair.

rtorrente, 2023-11-27:
Si je supprime totalement l'appareil sur home assistant et que je relance le container (en dernière version). L'appareil se recrée mais 

- Les id d'entités sont différents : teleinfo_hidden_bbrhpjb -> linky_hidden_bbrhpjb (Je pense que ce n'est pas la cause, c'est surtout que mes id n'étaient pas tout jeunes)
- Il manque des entités surtout (par exemple PTEC, PAPP, ETC)

[image]

En espérant que cela puisse t'aider à debug, n'hésites pas si besoin de plus de précisions

fmartinou, 2023-11-28:
> En creusant plus je me rend compte qu'il a été créé plein de nouvelles entitées qui ne correspondent pas à mes entitée

Oui, c'est une des modifs qui a été apportée.
C'est à dire qu'on créé tous les sensors qui correspondent au mode de téléinfo (historique ou standard) quel que soit l'offre edf (tempo, base...)
Les quelques sensors non utilisés par l'offre en cours resteront à "unknown".

Cela permet d'avoir une découverte fonctionnelle côté HA même chez les utilisateurs qui ont fréquemment des trames corrompues.

fmartinou, 2023-11-28:
Si je rollback en `8.3.3`, j'ai mes id qui sont également foirés car HA concatène les id du "device" (linky_$adco) et les id du sensor ("teleinfo_$adco_xxx")

Ce qui donne des id de sensors différents et à rallonge ()

La payload de découverte est :
```json
{
   "unique_id":"teleinfo_0123456789_ADCO",
   "name":"Teleinfo 0123456789 ADCO",
   "state_topic":"teleinfo/021861500433",
   "value_template":"{% if 'ADCO' in value_json %}{{ value_json.ADCO.raw }}{% else %}''{% endif %}",
   "device":{
      "identifiers":[
         "0123456789"
      ],
      "manufacturer":"Enedis",
      "model":"linky_0123456789",
      "name":"Linky 0123456789"
   }
}
```

Et HA me donne le unique_id suivant `sensor.linky_0123456789_teleinfo_0123456789_adco`

Vous avez le même comportement ?
Est-ce que HA ne s'est pas mis à définir des unique_id différement à partir d'une certaines version ?

pierrepinon, 2023-11-28:
Pour ma part, j'ai rollback avec une sauvegarde toute mon instance, car effectivement c'était un peu la foire

Il me semble que HA a changé des choses, mais je n'arrive plus à trouver l'info. Je tiens au courant

rtorrente, 2023-11-28:
Je me demande si ce n'est pas le même problème qu'observé ici sur le changement de nom dans HA cet été : https://github.com/Koenkk/zigbee2mqtt/issues/18445

Plus particulierement ce changement

J'ai remarqué que le unique_id est différent de l'entity_id (au moins dernièrement). Le unique_id n'est utilisé qu'en interne sur home assistant et référence toujours la même entité. 
C'est pour ça que je n'ai vu les différents nom qu'en supprimant complètement l'appareil.

fmartinou, 2023-11-30:
J'ai fait des modifs pour avoir les id des sensors sous la forme `sensor.teleinfo_$id-compteur_$etiquette`.
Par exemple : `sensor.teleinfo_0123456789_bbrhpjw`

Vous pouvez tester la version `fmartinou/teleinfo2mqtt:fix__70_break_entity_id` et me faire un retour, svp ?

rtorrente, 2023-11-30:
Salut @fmartinou 

Cela semble ok chez moi. Après avoir complètement supprimé l'appareil Teleinfo de HA. J'ai lancé le container sur la nouvelle version et les entités sont bien arrivées avec leur ancien nom (et sans le nom à rallonge).

Les entités manquantes de mon dernier post sont aussi bien présentes. 

Je pense qu'on est ok !

fmartinou, 2023-11-30:
Merci pour ton feedback rapide. 👍
Je vais releaser une version fix dans la journée du coup 😀