fmartinou/teleinfo2mqtt #74: Problème lors du discovery des sensors

stephanlascar, 2023-12-02:
Certaines données ne sont pas remonté dans home assistant. Après installation multiple, tout en prenant soin de tout initialiser, j'ai commencé à regarder un peu plus en détail.

Tout d'abord, j'ai une installation linky avec le téléinfo en mode standard, et en mode production activé via mon fournisseur.
J'utilise un dongle teleinfo USB de marque CGE electronic, branché en filaire directement sur Home Assistant en USB.

Via MQTT Explorer, j'ai d'abord remarqué que tout les noeuds ne sont pas publiés pour le discovery automatique. Dans le journal, je vois pourtant bien que le plugin publie tout les noeuds téléfinfo

> 23:10:20.994  INFO teleinfo2mqtt: Connected to MQTT broker [mqtt://homeassistant:1883]
23:10:24.424  INFO teleinfo2mqtt: Publish configuration for tag ADSC for discovery to topic [homeassistant/sensor/teleinfo/841875058573_adsc/config]
23:10:24.432  INFO teleinfo2mqtt: Publish configuration for tag CCAIN for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ccain/config]
23:10:24.433  INFO teleinfo2mqtt: Publish configuration for tag CCAIN-1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ccain-1/config]
23:10:24.434  INFO teleinfo2mqtt: Publish configuration for tag CCASN for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ccasn/config]
23:10:24.436  INFO teleinfo2mqtt: Publish configuration for tag CCASN-1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ccasn-1/config]
23:10:24.438  INFO teleinfo2mqtt: Publish configuration for tag DATE for discovery to topic [homeassistant/sensor/teleinfo/841875058573_date/config]
23:10:24.439  INFO teleinfo2mqtt: Publish configuration for tag DPM1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_dpm1/config]
23:10:24.440  INFO teleinfo2mqtt: Publish configuration for tag DPM2 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_dpm2/config]
23:10:24.442  INFO teleinfo2mqtt: Publish configuration for tag DPM3 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_dpm3/config]
23:10:24.445  INFO teleinfo2mqtt: Publish configuration for tag EAIT for discovery to topic [homeassistant/sensor/teleinfo/841875058573_eait/config]
23:10:24.448  INFO teleinfo2mqtt: Publish configuration for tag EASD01 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easd01/config]
23:10:24.450  INFO teleinfo2mqtt: Publish configuration for tag EASD02 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easd02/config]
23:10:24.451  INFO teleinfo2mqtt: Publish configuration for tag EASD03 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easd03/config]
23:10:24.454  INFO teleinfo2mqtt: Publish configuration for tag EASD04 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easd04/config]
23:10:24.455  INFO teleinfo2mqtt: Publish configuration for tag EASF01 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf01/config]
23:10:24.456  INFO teleinfo2mqtt: Publish configuration for tag EASF02 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf02/config]
23:10:24.458  INFO teleinfo2mqtt: Publish configuration for tag EASF03 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf03/config]
23:10:24.459  INFO teleinfo2mqtt: Publish configuration for tag EASF04 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf04/config]
23:10:24.460  INFO teleinfo2mqtt: Publish configuration for tag EASF05 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf05/config]
23:10:24.461  INFO teleinfo2mqtt: Publish configuration for tag EASF06 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf06/config]
23:10:24.462  INFO teleinfo2mqtt: Publish configuration for tag EASF07 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf07/config]
23:10:24.462  INFO teleinfo2mqtt: Publish configuration for tag EASF08 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf08/config]
23:10:24.463  INFO teleinfo2mqtt: Publish configuration for tag EASF09 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf09/config]
23:10:24.465  INFO teleinfo2mqtt: Publish configuration for tag EASF10 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_easf10/config]
23:10:24.466  INFO teleinfo2mqtt: Publish configuration for tag EAST for discovery to topic [homeassistant/sensor/teleinfo/841875058573_east/config]
23:10:24.467  INFO teleinfo2mqtt: Publish configuration for tag ERQ1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_erq1/config]
23:10:24.468  INFO teleinfo2mqtt: Publish configuration for tag ERQ2 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_erq2/config]
23:10:24.469  INFO teleinfo2mqtt: Publish configuration for tag ERQ3 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_erq3/config]
23:10:24.470  INFO teleinfo2mqtt: Publish configuration for tag ERQ4 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_erq4/config]
23:10:24.470  INFO teleinfo2mqtt: Publish configuration for tag FPM1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_fpm1/config]
23:10:24.471  INFO teleinfo2mqtt: Publish configuration for tag FPM2 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_fpm2/config]
23:10:24.473  INFO teleinfo2mqtt: Publish configuration for tag FPM3 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_fpm3/config]
23:10:24.474  INFO teleinfo2mqtt: Publish configuration for tag IRMS1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_irms1/config]
23:10:24.475  INFO teleinfo2mqtt: Publish configuration for tag IRMS2 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_irms2/config]
23:10:24.476  INFO teleinfo2mqtt: Publish configuration for tag IRMS3 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_irms3/config]
23:10:24.476  INFO teleinfo2mqtt: Publish configuration for tag LTARF for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ltarf/config]
23:10:24.477  INFO teleinfo2mqtt: Publish configuration for tag MSG1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_msg1/config]
23:10:24.478  INFO teleinfo2mqtt: Publish configuration for tag MSG2 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_msg2/config]
23:10:24.479  INFO teleinfo2mqtt: Publish configuration for tag NGTF for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ngtf/config]
23:10:24.480  INFO teleinfo2mqtt: Publish configuration for tag NJOURF for discovery to topic [homeassistant/sensor/teleinfo/841875058573_njourf/config]
23:10:24.480  INFO teleinfo2mqtt: Publish configuration for tag NJOURF+1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_njourf+1/config]
23:10:24.481  INFO teleinfo2mqtt: Publish configuration for tag NTARF for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ntarf/config]
23:10:24.482  INFO teleinfo2mqtt: Publish configuration for tag PCOUP for discovery to topic [homeassistant/sensor/teleinfo/841875058573_pcoup/config]
23:10:24.483  INFO teleinfo2mqtt: Publish configuration for tag PJOURF+1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_pjourf+1/config]
23:10:24.484  INFO teleinfo2mqtt: Publish configuration for tag PPOINTE for discovery to topic [homeassistant/sensor/teleinfo/841875058573_ppointe/config]
23:10:24.485  INFO teleinfo2mqtt: Publish configuration for tag PREF for discovery to topic [homeassistant/sensor/teleinfo/841875058573_pref/config]
23:10:24.486  INFO teleinfo2mqtt: Publish configuration for tag PRM for discovery to topic [homeassistant/sensor/teleinfo/841875058573_prm/config]
23:10:24.487  INFO teleinfo2mqtt: Publish configuration for tag RELAIS for discovery to topic [homeassistant/sensor/teleinfo/841875058573_relais/config]
23:10:24.488  INFO teleinfo2mqtt: Publish configuration for tag SINSTI for discovery to topic [homeassistant/sensor/teleinfo/841875058573_sinsti/config]
23:10:24.489  INFO teleinfo2mqtt: Publish configuration for tag SINSTS for discovery to topic [homeassistant/sensor/teleinfo/841875058573_sinsts/config]
23:10:24.489  INFO teleinfo2mqtt: Publish configuration for tag SINSTS1 for discovery to topic [homeassistant/sensor/teleinfo/841875058573_sinsts1/config]
[... cut ...]
Le noeud suivant est "NJOURF+1". _Ce pourrait t'il que le broker n'aime pas les noeuds contenant un "+" ? (Je crois avoir lu quelque chose là dessus...)_

Via MQTT Explorer, je me suis amusé à publié moi même un noeud en discovery: "SINSTI". Je vois bien directement celui-ci apparaitre dans le broker et il s'affiche immédiatement dans Home Assistant.

Je pense donc que les noeuds téléinfo contenant un caractère spécial ne semblent pas compatible... Mais ce ne sont que des suppositions....

Est il possible de renommer certains noeuds ? Ou d'ajouter une option dans la configuration pour ne publiquer que les noeuds qui nous interesse ?

stephanlascar, 2023-12-02:
Je viens de voir qu'il y a déjà une issue dessus; je vais attendre patiemment la correction ! Merci pour votre travail

closed 2023-12-02 by stephanlascar

fmartinou, 2023-12-09:
Fixed as `9.1.3`