fmartinou/teleinfo2mqtt #75: Entité manquante dans HA ?

fredarro, 2023-12-03:
Je suis en mode Standard.
Est-il normal de ne pas avoir la valeur SINSTS dans HA alors que je la vois bien sur mon broker MQTT ?

fredarro, 2023-12-04:
Oui tout est bien présent dans la trame mais dans HA plus rien à partir NJOURF
C'est à dire que les entités suivantes n'apparaissent pas dans HA :
NJOURF+1 
NTARF 
PCOUP 
PJOURF+1 
PPOINTE 
PREF 
PRM 
RELAIS 
SINSTI 
SINSTS 
SINSTS1 
SINSTS2 
SINSTS3 
SMAXIN 
SMAXIN-1 
SMAXSN 
SMAXSN-1 
SMAXSN1 
SMAXSN1-1 
SMAXSN2 
SMAXSN2-1 
SMAXSN3 
SMAXSN3-1 
STGE 
UMOY1 
UMOY2 
UMOY3 
URMS1 
URMS2 
URMS3 
VTIC

fmartinou, 2023-12-04:
Il y a des warnings ou des erreurs dans les logs du module ?

fredarro, 2023-12-04:
Non, rien Pas d'erreur ou de warning
Toutes les entités arrivent bien sur le Broker MQTT.
On dirait une limitation à 40 entités remontées de MQTT vers HA par ordre alphabétique.

fmartinou, 2023-12-04:
C'est potentiellement le même problème que https://github.com/fmartinou/teleinfo2mqtt/issues/16#issuecomment-1817828624.

Le  problème avec un `+` dans le nom du topic qui n'est pas valide d'un point de vue `mqtt`.
Ce qui expliquerait pourquoi c'est à partir de `NJOURF+1`, que ça ne remonte plus 🤔 .

Faut que je prenne le temps de renommer tous ces `+`

fredarro, 2023-12-04:
Super, tu as surement raison et le `+` doit être à l'origine du problème.
Nous attendrons le renommage.

spamoi33, 2023-12-04:
Hello @fmartinou , tu dis "d'un point de vue mqtt" mais pourtant les valeurs sont OK quand on regarde avec un "explorateur" mqtt. C'est HA qui n'arrive pas à lire les infos du broker du coup ?

fmartinou, 2023-12-05:
Les données publiées au format JSON dans le topic mqtt `teleinfo/$adsc` ne posent pas de problème car les `+` sont juste des caractères apparaissant dans le nom des clés du document JSON ; c'est parfaitement valide.

Le problème vient probablement plutôt des topics qui servent à publier la configuration des sensors dans HA (pour la découverte automatique des sensors par HA).
Ce sont les topics apparaissant sous ` homeassistant/sensor/teleinfo/$adsc_$label/config`)

Ces topics, que le module cherche à créer, contiennent des `+` dans leurs noms pour les étiquettes `NJOURF+1`...
Exemple  :
```
 homeassistant/sensor/teleinfo/0123456789_njourf+1/config
```

or le `+` a une signification en mqtt (c'est un wildcard) donc je suppose que le problème vient de là.

stephanlascar, 2023-12-06:
J'avais regardé dans mqttexplorer, en effet les noeuds à partir du NJOURF+1 (inclu) ne sont pas publié dans le discovery

ThibautBascoulergue, 2023-12-06:
Bonjour,
Est-ce qu'on a une solution pour ce problème, depuis la dernière mise à jours de HA je n'ai plus certaines valeurs comme évoqué au-dessus.
Merci

Balakao, 2023-12-06:
> Bonjour, Est-ce qu'on a une solution pour ce problème, depuis la dernière mise à jours de HA je n'ai plus certaines valeurs comme évoqué au-dessus. [...]

Salut,

De mon côté, j'ai filtré les entités par sensor.linky, et j'ai fait un par un pour chaque compteur :
- récupération de l'ID du compeur
- suppression du compteur obsolète (ça ne supprime pas l'historique)
- renommage du nouveau compteur avec l'id de l'ancien

Ex : sensor.linky_012345678910_teleinfo_012345678910_bbrhcjw (nom de l'ancien compteur), on récupère le nom sans sensors., on va sur le nouveau compteur sensor.linky_012345678910_bbrhcjw, et on remplace linky_012345678910_bbrhcjw par sensor.linky_012345678910_teleinfo_012345678910_bbrhcjw.

Ça permet de remettre à flot sans avoir à tout reconfigurer et perdre les historiques. Penser à bien faire une sauvegarde avant, modifier, et faire un reboot pour bien s'assurer que ça fonctionne sans surprise (quand on fait un reboot 3 semaines après, on a tendance à ne plus faire le lien entre update et conneries).

ThibautBascoulergue, 2023-12-06:
Ok , merci

fmartinou, 2023-12-09:
Fixed as `9.0.3`

closed 2023-12-09 by fmartinou

spamoi33, 2023-12-10:
Merci pour la correction @fmartinou  :)
Bon, c'est juste un peu pénible de devoir tout "recabler" derrière, mais au moins ça fonctionne ^^

Par curiosité, pourquoi certains sensors sont dupliqués ? exemple, j'ai 3 SINSTS (sinsts1 => 3)

Bon dimanche !

fmartinou, 2023-12-10:
Merci pour le feedback.
Les 3 sinsts font partie de la spécification du protocole en mode standard.
Je pense que c'est pour les installations en triphasé.
Cdt

spamoi33, 2023-12-10:
Ah ok, je ne suis pas en triphasé pourtant. Ca ne gêne pas plus que ça.

fmartinou, 2023-12-10:
> Ah ok, je ne suis pas en triphasé pourtant. Ca ne gêne pas plus que ça.

Tous les sensors du mode standard sont automatiquement créés quels que soient l'abonnement et les caractéristiques du compteur (monophasé, triphasé...).

Voir [Enedis-NOI-CPT_54E](https://www.enedis.fr/media/2035/download)[](https://www.enedis.fr/media/2035/download) pour avoir la liste exhaustive.

spamoi33, 2023-12-10:
Effectivement... ça ne m'avait sauté aux yeux 👀

spamoi33, 2023-12-12:
Hello,

J'ai donc modifier les sources de mon dashboard Energie, du coup forcement j'ai perdu tous les historiques de ma consommation :|

Quelqu'un aurais une idée pour transférer les données ? ❤️

spamoi33, 2023-12-15:
Personne ? :( je n'ai plus aucune consommation sur le dashboard energy avant le problème de nom de topic MQTT :'(

Comment faites vous ??