fmartinou/teleinfo2mqtt #76: ECONNRESET erreur depuis une semaine  non expliquée

eetasse, 2023-12-06:
Hello, tout baigne depuis des mois et subitement plus de valeurs TeleInfo dans HA. Ça tourne sur HA os sur raspberry 4. J'ai joué avec la config teleinfo2mqqt, la config MQTT mosquito, désinstaller et réinstaller ces modules mais rien n'y fait pb de econnreset toujours présent et pas d'infos envoyées à mqtt. J'ai beau chercher sur le net je sèche... hellpppppp

:56.997  INFO teleinfo2mqtt: Publish configuration for tag URMS3 for discovery to topic [homeassistant/sensor/teleinfo/061764362745_urms3/config]
18:16:56.998  INFO teleinfo2mqtt: Publish configuration for tag VTIC for discovery to topic [homeassistant/sensor/teleinfo/061764362745_vtic/config]
18:16:57.115  WARN teleinfo2mqtt: Error when publishing to the mqtt broker (read ECONNRESET)
18:16:58.119  INFO teleinfo2mqtt: Reconnecting to the MQTT broker...

Arbassan, 2023-12-09:
bonjour ,
je crois bien que j'ai la même depuis ce matin , j'ai cru que c'était la mise à jour de HA mais visiblement pas, sous mqtt j'ai bien les données qui arrivent 
homeassistant/sensor/teleinfo/022062053230_vtic/config]
10:45:36.511  WARN teleinfo2mqtt: Error when publishing to the mqtt broker (read ECONNRESET)
10:45:37.513  INFO teleinfo2mqtt: Reconnecting to the MQTT broker...
11:25:12.691  WARN teleinfo2mqtt: Invalid value received for label EASD04 [00158450EAIT]

Makokoy79, 2023-12-09:
Salut à tous,
Mon intégration a évoluée depuis la dernière version en 9.0.3. J'étais bloqué sur PJOURF+1.
Je suis maintenant sur la même erreur que mentionnée ci-dessus : 
```
17:56:12.005  INFO teleinfo2mqtt: Publish configuration for tag VTIC for discovery to topic [homeassistant/sensor/teleinfo2mqtt/041864289085_vtic/config]
21:02:07.486  INFO teleinfo2mqtt: Reconnecting to the MQTT broker...
21:02:08.496  INFO teleinfo2mqtt: Reconnecting to the MQTT broker...
21:02:08.499  WARN teleinfo2mqtt: Error when publishing to the mqtt broker (read ECONNRESET)
```

On avance, merci pour le boutot @fmartinou ! :-)

eetasse, 2023-12-17:
Bonjour 
J'ai rétrogradé à la version 8.3.3 de teleinfo2mqtt et ça fonctionne très bien.