fmartinou/teleinfo2mqtt #78: Home Assistant Configuration

guigeek34, 2023-12-27:
Bonjour,
Il ne s'agit pas d'un bug, désolé, mais j'aurais besoin d'aide pour avancer...
Pourriez-vous partager l'extrait votre configuration.yml concernant Home Assistant, svp?
Merci d'avance

fmartinou, 2024-02-04:
Bonjour,

Vous parlez de la configuration de l'add-on ?

Il n'y a pas beaucoup de conf à définir ; la plupart des valeurs par défaut sont ok.

En prérequis, il vous faut un Home-Assistant et un Broker MQTT qui fonctionnent.

Et ensuite, les seules options à surcharger sont :
### MQTT_URL
La valeur `mqtt://homeassistant:1883` est en principe ok si on installer mosquitto avec l'addon officiel fourni avec home-assistat

### MQTT_USER et MQTT_PASSWORD
Dépend de si vous avez activé l'authentification sur votre broker mosquitto.

### SERIAL
La valeur `/dev/ttyUSB0` est en principe ok mais peut dépendre du point de montage exact du module USB téléinfo

guigeek34, 2024-03-03:
Bonjour,
Finalement j'avais trouvé par moins-même les infos nécessaires.
Au cas où ça peu en aider d'autres :
```
mqtt:
  sensor:
    - name: "index conso"
      unique_id: index_conso
      state_topic: "teleinfo/conso"
      device_class: energy
      state_class: total_increasing
      unit_of_measurement: "Wh"
      value_template: '{{ value_json["BASE"]}}'
    - name: "Puissance apparente "
      unique_id: papp
      state_topic: "teleinfo/conso"
      device_class: energy
      state_class: measurement
      unit_of_measurement: "W"
      value_template: '{{ value_json["PAPP"]}}'
```

guigeek34, 2024-03-03:
Example in previous post

closed 2024-03-03 by guigeek34