fmartinou/teleinfo2mqtt #80: Teleinfo2mqtt: pas de remontée HP/HC

sda17, 2024-02-14:
J’ai un contrat HP/HC mais les infos remontées sont celles d’un contrat de base

## Ma configuration teleinfo2mqtt
EMIT_INTERVAL: 10
HASS_DISCOVERY_PREFIX: homeassistant
LOG_LEVEL: info
MQTT_BASE_TOPIC: teleinfo
MQTT_URL: mqtt://core-mosquitto:1883
MQTT_USER: homeassistant
MQTT_PASSWORD: homeassistant
SERIAL: /dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DN04T0HG-if00-port0
TIC_MODE: history
TEMPO_ENABLED: false
ADAPTER: ezsp

Le lancement du module teleinfo2mqtt ne me remonte pas d’erreur particuliere

18:01:43.831 INFO teleinfo2mqtt: Connected to port [/dev/serial/by-id/usb-FTDI_FT230X_Basic_UART_DN04T0HG-if00-port0]
18:01:43.833 INFO teleinfo2mqtt: Connecting to MQTT broker [mqtt://core-mosquitto:1883]
18:01:43.950 WARN teleinfo2mqtt: Invalid value received for label 864037122 [?]
18:01:44.159 INFO teleinfo2mqtt: Connected to MQTT broker [mqtt://core-mosquitto:1883]

Je n’ai aucune remontée sur la partie HP/HC et même le PTEC n’est pas correct
[image]
[image]

C’est comme-ci, mon contrat est de type « base » alors que je reçois bien l’impulsion d’ENGIE heures creuses 2 fois par jour.
[image]

Merci pour votre aide.

fmartinou, 2024-02-16:
Pouvez-vous relancer avec les logs en `debug` ?
Pour voir s'il y a des éléments qui peuvent expliquer.

sda17, 2024-02-16:
Les voici
[9afc8f77_teleinfo2mqtt_2024-02-16T19-41-00.432Z.log](https://github.com/fmartinou/teleinfo2mqtt/files/14315330/9afc8f77_teleinfo2mqtt_2024-02-16T19-41-00.432Z.log)

sda17, 2024-02-16:
Sous jeedom, j'avais bien les deux compteurs :(

sda17, 2024-02-18:
[9afc8f77_teleinfo2mqtt_2024-02-18T11-54-39.583Z.log](https://github.com/fmartinou/teleinfo2mqtt/files/14322323/9afc8f77_teleinfo2mqtt_2024-02-18T11-54-39.583Z.log)

sda17, 2024-05-29:
Suite à la mise à jour de homeassistant, j'ai récupéré les index hp et hc ainsi que le ptec
Core
2024.5.5
Supervisor
2024.05.1
Operating System
12.3
Frontend
20240501.1

fmartinou, 2024-06-12:
OK :man_shrugging: 
Je peux donc cloturer ?

777-Bogart, 2024-12-04:
Bonjour,

J'ai le même problème : HCHC & HPHP ne sont pas valorisés.
cf les PJ (screenshot + logs).

Mon installation est très récente :
 - HA :
Core 2024.11.3
Supervisor 2024.11.4
Operating System 13.2
Interface utilisateur 20241106.2
 - teleinfo2mqtt : 9.0.4

module téléinfo "GCE electronics" : https://www.gce-electronics.com/en/-usb/655-module-teleinfo-usb.html

[image]
[log.txt](https://github.com/user-attachments/files/18008428/log.txt)

Merci pour ton aide.

777-Bogart, 2024-12-05:
J'ai trouvé le "problème" : les informations sont dans "BASE" (et non HCHC/HPHP)
13:02:58.774 DEBUG teleinfo2mqtt: Split frame [BASE,001330675,$,]