fmartinou/teleinfo2mqtt #81: Passage en mode standard et suppression des anciennes valeurs du mode historique

SmartHbdev, 2024-03-02:
Bonjour,

Nouveau dans le mode Home assistant, j'étais en mode historique quelques jours.
Ensuite je suis passé en mode standard, tout va bien, sauf que j'ai les variables des deux modes maintenant.

Comment faire pour supprimer tout du mode historique pour ne pas polluer la liste des entités ?
Désinstaller le module teleinfo2mqtt puis le reinstaller ne change rien du tout.

A part la solution de cacher un par un les entités, je ne trouve pas. Et ça me parait long comme solution.

fmartinou, 2024-03-05:
Bonjour,

Avec un client mqtt ([Mqtt Explorer](https://mqtt-explorer.com/) par ex), tu peux supprimer tous les topics des entités que tu ne veux plus voir.

[image]

Par contre, je ne sais pas si HA va garder les "long-term stats" ; càd continuer à afficher les valeurs des anciens sensors dans la partie "Energy" 🤷‍♂️

SmartHbdev, 2024-03-08:
Mais MQTT explorer permet de regarder les quelques dernières valeurs et celles en quasi temps réel. Je ne vois pas trop comment on pourrait modifier des valeurs des entités de plusieurs jours ou semaines en arrières.
Il faudrait peut être voir où Home assistant garde ça. Je ne sais pas si c'est la bonne voie à suivre vu que je suis débutant