fmartinou/teleinfo2mqtt #83: 2 Usb Devices

jacky017190, 2024-04-10:
Hello, 
how do I connect 2 USB devices (2 meters) to teleinfo? I have 2 devices:
/dev/serial/by-id/usb-CH2i_uTinfo-V3.0_TINFO-2379-if00
/dev/serial/by-id/usb-CH2i_uTinfo-V3.0_TINFO-2263-if00

When I configure the first device, I receive the informations, but I cannot add the second.

Thanks

jlegall56, 2024-08-31:
Hello, same question for me.
With particularity i have the first on standard mode ( Connect on linky), and the second in history mode ( a personnal device) for my solar installation.

Thanks a lot
Julien.

fmartinou, 2024-09-04:
Hi,

The service is intended to read 1 device only.

If you need to receive data from multiple devices, then you have to run multiple teleinfo2mqtt instances (one per device).

e.g.

```yaml
version: '3'

services:
  teleinfo2mqtt1:
    image: fmartinou/teleinfo2mqtt
    devices:
      - /dev/serial/by-id/my-usb-device-1:/dev/ttyUSB0
    environment:
      - MQTT_URL=...

  teleinfo2mqtt2:
    image: fmartinou/teleinfo2mqtt
    devices:
      - my-usb-device-2:/dev/ttyUSB0
    environment:
      - MQTT_URL=...
```

The only limitation is if you run it from hassio because hassio prevents from running an addon multiple times (if I'm not wrong).