fmartinou/teleinfo2mqtt #88: Perte des informations TEMPO

EMqA44, 2024-08-30:
Bonjour.
Depuis quelques jours, il n'y a plus de remontée des infos tempo. Cela semble lié à la méthode de récupération de ces infos chez EDF car d'autres méthodes semblent également touchées.
https://forum.hacf.fr/t/api-pour-recuperer-tempo-edf/12156/21
Cordialement.

```
19:40:55.042  INFO teleinfo2mqtt: Connecting to port [/dev/serial/by-id/usb-CH2i_uTinfo-V3.0_TINFO-2201-if00] with history TIC mode
19:40:55.045  INFO teleinfo2mqtt: Connected to port [/dev/serial/by-id/usb-CH2i_uTinfo-V3.0_TINFO-2201-if00]
19:40:55.046  INFO teleinfo2mqtt: Connecting to MQTT broker [mqtt://core-mosquitto:1883]
19:40:55.119  INFO teleinfo2mqtt: Connected to MQTT broker [mqtt://core-mosquitto:1883]
19:40:55.120  INFO teleinfo2mqtt: Start polling tempo data
19:40:55.283 ERROR teleinfo2mqtt: Error when fetching tempo data (Request failed with status code 404)
19:41:56.640  INFO teleinfo2mqtt: Publish configuration for tag ADCO for discovery to topic [homeassistant/sensor/teleinfo2mqtt/032161005388_adco/config]
19:41:56.641  INFO teleinfo2mqtt: Publish configuration for tag ADIR1 for discovery to topic [homeassistant/sensor/teleinfo2mqtt/032161005388_adir1/config]
19:41:56.641  INFO teleinfo2mqtt: Publish configuration for tag ADIR2 for discovery to topic [homeassistant/sensor/teleinfo2mqtt/032161005388_adir2/config]
```

morganzion, 2024-08-31:
same problem here :/

romainrossi, 2024-09-03:
Je confirme, même problème également :

`ERROR teleinfo2mqtt: Error when fetching tempo data (Request failed with status code 404)`

fmartinou, 2024-09-03:
Je regarde pour corriger ça.

fmartinou, 2024-09-03:
J'ai fait un fix pour basculer sur l'API https://www.api-couleur-tempo.fr.

Ca permet de retrouver la couleur du jour et du lendemain.

Par contre, cette API ne remonte pas le nombre de jours consommés / restants par couleur sur la saison ; pour cela il faudra patienter un peu qu'une solution se dégage.

Je vais releaser une version `9.0.4`

closed 2024-09-03 by fmartinou

fmartinou, 2024-09-03:
Version `9.0.4` dispo.

adicorra, 2024-09-05:
Ok merci pour le fix et bien noté. Cependant je ne comprends pas bien le rapport avec les serveurs enedis. Dans la spec TIC sont bien prévu 11 paramètres liés au Tempo (cf en page 24 registre des statuts). Ils ne sont plus sur la trame ? Est-ce temporaire / pb technique ou une décision de leur part de ne plus le communiquer ?

fmartinou, 2024-09-05:
Les infos tempo sur les couleurs du jour/lendemain et le nombre de jours restants par couleur sur la saison étaient récupérées via une API HTTP, pas depuis la téléinfo.

adicorra, 2024-09-05:
Ah d'accord merci, j'apprends cela. J'ai vu au passage que sur le TIC il y a un champ "DEMAIN" mais uniquement en mode historique.