fmartinou/teleinfo2mqtt #90: Décodage STGE et couleur du lendemain

picsou83, 2024-09-13:
Je souhaiterais demander l'ajout de la fonctionnalité de décodage du registre STGE pour le compteur Linky au sein du module teleinfo2mqtt. Le registre STGE fournit des informations précieuses sur l'état du compteur et surtout la couleur du lendemain via la TIC.

Explication détaillée :
Le registre STGE (registre des statuts en mode standard) est composé de 4 octets encodés en hexadécimal qui fournissent divers états opérationnels du compteur Linky. Chaque octet contient des informations spécifiques telles que le dépassement de puissance, les détails tarifaires, le sens de l'énergie, et les états de communication.

<article class="w-full text-token-text-primary focus-visible:outline-2 focus-visible:outline-offset-[-4px]" dir="auto" data-testid="conversation-turn-17" data-scroll-anchor="false"><div class="text-base py-[18px] px-3 md:px-4 m-auto w-full md:px-5 lg:px-1 xl:px-5"><div class="mx-auto flex flex-1 gap-4 text-base md:gap-5 lg:gap-6 md:max-w-3xl lg:max-w-[40rem] xl:max-w-[48rem]"><div class="group/conversation-turn relative flex w-full min-w-0 flex-col agent-turn"><div class="flex-col gap-1 md:gap-3"><div class="flex max-w-full flex-col flex-grow"><div data-message-author-role="assistant" data-message-id="a661a082-7be9-415d-af4d-eeaf0a57f39c" dir="auto" class="min-h-[20px] text-message flex w-full flex-col items-end gap-2 whitespace-normal break-words [.text-message+&amp;]:mt-5"><div class="flex w-full flex-col gap-1 empty:hidden first:pt-[3px]"><div class="markdown prose w-full break-words dark:prose-invert dark"><p>Exemple avec le code <strong>013A0000</strong> composé de 4 octets (en hexadécimal) :</p><ul><li><strong>Octet 4 : 01</strong></li><li><strong>Octet 3 : 3A</strong></li><li><strong>Octet 2 : 00</strong></li><li><strong>Octet 1 : 00</strong></li></ul><h3>Conversion hexadécimale → binaire :</h3><ol><li><strong>01</strong> (hexadécimal) → <strong>00000001</strong> (binaire)</li><li><strong>3A</strong> (hexadécimal) → <strong>00111010</strong> (binaire)</li><li><strong>00</strong> (hexadécimal) → <strong>00000000</strong> (binaire)</li><li><strong>00</strong> (hexadécimal) → <strong>00000000</strong> (binaire)</li></ol><h4>Code binaire complet :</h4><pre class="!overflow-visible"><div class="dark bg-gray-950 contain-inline-size rounded-md border-[0.5px] border-token-border-medium relative"><div class="flex items-center text-token-text-secondary bg-token-main-surface-secondary px-4 py-2 text-xs font-sans justify-between rounded-t-md h-9"></div><div class="sticky top-9 md:top-[5.75rem]"><div class="absolute bottom-0 right-2 flex h-9 items-center"><div class="flex items-center rounded bg-token-main-surface-secondary px-2 font-sans text-xs text-token-text-secondary"><span class="" data-state="closed"></span></div></div></div><div class="overflow-y-auto p-4" dir="ltr"><code class="!whitespace-pre hljs">00000001 00111010 00000000 00000000
</code></div></div></pre>Voici le décodage et l'interprétation pour chaque octet :

**Octet 1 : 00 (binaire : 00000000)**
Cet octet fournit des informations sur le dépassement de puissance, la surtension, le cache-borne et l'organe de coupure.

Dépassement de puissance : pas de dépassement (valeur 0).
Surtension : pas de surtension sur une phase (valeur 0).
Cache-borne distributeur : fermé (valeur 0).
Organe de coupure : fermé (valeur 0, aucune coupure).

**Octet 2 : 00 (binaire : 00000000)**
Cet octet fournit des informations sur les tarifs en cours et le sens de l’énergie active.

Tarif distributeur : index 1 (valeur 00).
Tarif fournisseur : index 1 (valeur 0000).
Sens de l’énergie active : consommation d’énergie active (valeur 0).
Mode de fonctionnement : consommateur (valeur 0).

**Octet 3 : 3A (binaire : 00111010)**
Cet octet concerne la communication, la synchronisation CPL, et le statut de l'horloge.

Synchronisation CPL : hors synchronisation (valeur 0).
Statut CPL : enregistré (valeur 10).
Sortie com Euridis : activée sans sécurité (valeur 01).
Téléinformation : en mode standard (valeur 1).
Horloge interne : correcte (valeur 0).

**Octet 4 : 01 (binaire : 00000001)**
Cet octet contient des informations sur les pointes mobiles et les couleurs du contrat Tempo.

Pointe mobile : pas de pointe mobile (valeur 00).
Préavis de pointe mobile : pas de préavis (valeur 00).
Couleur Tempo du jour : bleu (valeur 01).
Couleur Tempo du lendemain : non définie, pas d'annonce (valeur 00).

Cas spécifique à la couleur du lendemain (Couleur Tempo) :
Il est important de noter que, lorsque la Couleur du jour est Bleu (01 dans les bits 25..24), la Couleur du lendemain est  comme 00 (aucune annonce). Dans ce cas, il serait souhaitable que le décodage reflète cette règle : la couleur du lendemain reste à 0 lorsqu'il n'y a pas d'annonce ou que la couleur du jour est bleu.

Quelques informations sur STGE : 
https://github.com/sosandroid/Monitoring_Energie/blob/main/doc/tic_standard_histo.md
et pour la couleur du lendemain à 00 si bleu:
http://hacks.slashdirt.org/sw/tic2json/

devchristof, 2024-09-24:
Je suis intéressé aussi par et fonctionnalité

romainrossi, 2024-12-29:
J'ai bricolé des entités templates pour interpretter le registre d'état STGE et produire différentes informations utiles. J'avais également dans l'idée d'utiliser ça au lieu de l'API pour obtenir la couleur du jour tempo, mais la mise à jour du STGE avec les annonces tempo est très tard dans la journée (20h environ); alors que l'appli smartphone EDF et l'API donnent la même info plus tôt dans la journée.

Néanmoins, si ça vous interresse, voici mes templates pour les couleurs du jour et de demain à partir du STGE :

```
binary_sensor:
  - name: stge_tempo_today_unanounced
    unique_id: template_teleinfo_stge_tempo_today_unanounced
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_today_unanounced = (stge // (2**24)) | bitwise_and(3) == 0b00 %}
      {{ tempo_today_unanounced }}
  - name: stge_tempo_today_blue
    unique_id: template_teleinfo_stge_tempo_today_blue
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_today_blue = (stge // (2**24)) | bitwise_and(3) == 0b01 %}
      {{ tempo_today_blue }}
  - name: stge_tempo_today_white
    unique_id: template_teleinfo_stge_tempo_today_white
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_today_white = (stge // (2**24)) | bitwise_and(3) == 0b10 %}
      {{ tempo_today_white }}
  - name: stge_tempo_today_red
    unique_id: template_teleinfo_stge_tempo_today_red
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_today_red = (stge // (2**24)) | bitwise_and(3) == 0b11 %}
      {{ tempo_today_red }}

  - name: stge_tempo_tomorrow_unanounced
    unique_id: template_teleinfo_stge_tempo_tomorrow_unanounced
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_tomorrow_unanounced = (stge // (2**26)) | bitwise_and(3) == 0b00 %}
      {{ tempo_tomorrow_unanounced }}
  - name: stge_tempo_tomorrow_blue
    unique_id: template_teleinfo_stge_tempo_tomorrow_blue
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_tomorrow_blue = (stge // (2**26)) | bitwise_and(3) == 0b01 %}
      {{ tempo_tomorrow_blue }}
  - name: stge_tempo_tomorrow_white
    unique_id: template_teleinfo_stge_tempo_tomorrow_white
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_tomorrow_white = (stge // (2**26)) | bitwise_and(3) == 0b10 %}
      {{ tempo_tomorrow_white }}
  - name: stge_tempo_tomorrow_red
    unique_id: template_teleinfo_stge_tempo_tomorrow_red
    state: >
      {% set stge = int(states('sensor.teleinfo_stge'), base=16) %}
      {% set tempo_tomorrow_red = (stge // (2**26)) | bitwise_and(3) == 0b11 %}
      {{ tempo_tomorrow_red }}

```

Les valeurs à utiliser pour les autres champs du registre :
```
{% set stge = int(states('sensor.teleinfo_stge'), base=16) %}

{%- set contact_sec = (stge // 2**0 | bitwise_and(1)) == 0 -%}
{%- set organe_de_coupure = (stge // 2**1 | bitwise_and(0b111)) == 0 -%}
{%- set etat_cache_borne_distributeur_closed = (stge | bitwise_and(0b10000)) == 0 -%}
{%- set overvoltage = (stge | bitwise_and(0b1000000) != 0) -%}
{%- set overload = (stge | bitwise_and(0b10000000) != 0) -%}
{%- set consuming = (stge | bitwise_and(0b100000000) == 0) -%}
{%- set producing = not consuming -%}
{%- set active_energy_positive = (stge | bitwise_and(0b1000000000)) == 0 -%}
{%- set idx_tarif_fournisseur = stge | bitwise_and(0x00003C00)//(2**10) -%}
{%- set idx_tarif_distributeur = stge | bitwise_and(0x0000C000)//(2**14) -%}
{%- set clock_ok = (stge | bitwise_and(2**16))==0 -%}
{%- set teleinfo_mode = ( stge // (2**17)) | bitwise_and(1) -%}
{%- set teleinfo_mode_history = ( stge // (2**17)) | bitwise_and(1) == 0 -%}
{%- set teleinfo_mode_standard = ( stge // (2**17)) | bitwise_and(1) == 1 -%}
{%- set euridis_com_enabled = (stge // (2**19)) | bitwise_and(1) == 1 -%}
{%- set euridis_com_secured = (stge // (2**20)) | bitwise_and(1) == 1 -%}
```

Je n'ai pas encore tout testé et l'écriture est améliorable...

closed 2025-06-03 by picsou83