fmartinou/teleinfo2mqtt #93: teleinfo2mqtt arrête d'envoyer des informations au bout de quelques minutes

vdanjean, 2024-11-15:
Bonjour,

Mon compteur est un linky en mode standard. J'ai acheté un convertisseur pour avoir les infos sur une ligne série. Avec minicom, je vois bien les bonnes infos arriver depuis l'USB en continu.
J'ai mis en place teleinfo2mqtt avec hassio. Quand je le lance, les infos correctes sont bien remontées. Les sensors affichent des info cohérentes. Mais, après quelques minutes (moins de 10 min, plutôt 1 ou 2), plus rien n'évolue (y compris sur plusieurs jours si je ne fais rien).
Ça reste comme cela jusqu'à ce que je relance le module teleinfo2mqtt (depuis l'interface web de HA). Dans ce cas, je récupère les nouvelles valeurs, ça marche quelques minutes, puis les relevés s'arrêtent.
Si je mets les logs en mode DEBUG, je vois bien les trames traitées, mais, au bout de quelques minutes, les messages s'arrêtent brutalement, sans aucun message différent des précédents.

Si je regarde le brocker mqtt, par exemple avec `mosquitto_sub -t '#' -u admin -P XXXX -v`, là encore, je vois bien des messages publiés sur teleinfo/XXXXXXX et teleinfo/status pendant quelques minutes, puis plus rien de teleinfo2mqtt (je continue bien de voir les updates de zigbee2mqtt par contre).

Avez-vous une idée de pouquoi teleinfo2mqtt s'arrête systématiquement de fonctionner au bout de quelques minutes sans laisser de trace dans les logs ? Des idées pour creuser un peu plus et récolter de l'info ?

Cordialement,

plop3, 2025-12-10:
J'ai trouvé la raison de mon problème: des micro-coupures EDF
Micro-coupure -> Plus de remontée d'infos jusqu'au redémarrage de teleinfo2mqtt

Braccac, 2025-12-10:
J'ai des panneaux solaires aussi mais comme le linky renvoie des volts ampères la valeur varie même si on est à zéro il y a toujours un équipement qui renvoie une puissance apparente

plop3, 2025-12-10:
J'ai créé un capteur mqtt TeleinfoStatus et une automatisation.
Si le status passe de "up" à autre chose, le module est redémarré.

plop3, 2025-12-19:
Je viens d'avoir une micro-coupure.
Mon automatisation a fonctionné et redémarré Teleinfo2mqtt \o/