fmartinou/teleinfo2mqtt #94: couleur DEMAIN

Braccac, 2024-12-15:
Bonjour,
sur la trame envoyer il y a bien de retour de la couleur de demain , extrait de la trame reçue:
20:26:31.099 DEBUG teleinfo2mqtt: Raw frame [DEMAIN ROUG +
]
20:26:31.099 DEBUG teleinfo2mqtt: Split frame [DEMAIN,ROUG,+,]
20:26:31.100 DEBUG teleinfo2mqtt: Value for label DEMAIN = ROUG

je sais pas si cela permet de remettre la valeur par un retour de l'info par teleinfo.

merci

Braccac, 2024-12-21:
je n'ai pas internet tout le temps et cela permettrai d'avoir la couleur de demain sans internet.
par exemple si la connexion n'est pas détecté, prendre en compte la teleinfo. du coup pour les automatismes ça marche encore si on prends la bonne entité

fmartinou, 2024-12-21:
Ça me semble être un cas trop particulier pour que je rajoute cela dans teleinfo2mqtt 🤔.
La gestion du fallback entite_demain_par_api -> entite_demain_par_teleinfo peut tout aussi bien être faite côté client je pense (dans home assistant par exemple), non ?

romainrossi, 2024-12-29:
@Braccac : j'avais la même idée, j'ai fait des entités templates pour obtenir cette information à partir du registre STGE disponible en mode standard (DEMAIN est uniquement disponible en mode historique).
J'ai mis mon code dans un commentaire sur une autre issue : https://github.com/fmartinou/teleinfo2mqtt/issues/90#issuecomment-2564862680

smuteau, 2025-02-03:
Hello, je ne sais pas si cette API pourrait aider (et si elle est toujours d'actualité), dans le doute :  
https://data.rte-france.eu/catalog/-/api/consumption/Tempo-Like-Supply-Contract/v1.1#
https://data.rte-france.eu/web/guest/catalog/-/api/doc/user-guide/Tempo+Like+Supply+Contract/1.1
PS : je ne suis pas dev mais le projet m'intéresse :)